PeterSuh-Q3/tinycore-redpill #4: server rebooted imediate after selecting grub menu 1

zfdang, 2023-08-25:
friend entry is ok, but for the first 2 entries, the server reboots immediately...

what could be the problem?

PeterSuh-Q3, 2023-08-26:
> friend entry is ok, but for the first 2 entries, the server reboots immediately...
> 
> what could be the problem?

Seeing your menu like this one?
If so, this JOT mode does not go through a separate GNU kernel for intermediate processing in the same sense as ARPL's Direct Boot.
JOT is the way it was originally developed and goes straight into synology kernel loading.

[image]

zfdang, 2023-08-26:
I'm able to install 7.2.0 successfully after using clean m-shell image. It works perfectly.
Previously, I just downloaded m-shell scripts with original tcrp image.

closed 2023-08-26 by zfdang