PeterSuh-Q3/tinycore-redpill #6: missing the /dev/dri system drivers folder?

erich1010, 2023-08-31:
Hello. When I try to locate the /dev/dri system drivers folder in the installed DSM, it cannot be found. I am not sure if this is supposed to be part of the install, but if these directories are not present, it seems like it is impossible to use transcoding in Jellyfin. I have tried using both the DS920+ and the DS1520+, both of which are supposed to be able to do transcoding, but both of them do not contain the /dev/dri folder which is required by Jellyfin. Is this something that should be in the loader? Sorry, I am not very knowledgeable with all the technical stuff here, was just hoping I could find a solution to this issue.

PeterSuh-Q3, 2023-09-02:
For transcoding, an Intel CPU between the 4th and 10th generations with a built-in iGPU must be used. 
AMD systems with a built-in iGPU are also not possible.

erich1010, 2023-09-02:
Ok, thanks.

closed 2023-10-01 by PeterSuh-Q3

koznov, 2025-01-23:
Hello @PeterSuh-Q3,

It is still not possible to transcode using AMD internal graphics? I'm wondering, because I've recently built system using your loader with version DS1621xs+ (DSM 7.2.2-72806 Update 2) on Motherboard based on Ryzen 7 8845HS with AMD Radeon 780M GPU inside.

PeterSuh-Q3, 2025-01-24:
Transcoding is limited to platforms using the i915 module based on Intel IGPU. 
Only DS918+ Apollo Lake, DS920+ Gemini Lake, SA6400 EPYC 7002 are possible.

PeterSuh-Q3, 2025-01-24:
> Hello [@PeterSuh-Q3](https://github.com/PeterSuh-Q3),
>
> It is still not possible to transcode using AMD internal graphics? I'm wondering, because I've recently built system using your loader with [...]

However, if you are interested in transcoding with Jellyfin and Ryzen, please refer to the guide below. It seems to be possible only on Linux, not on Synology.

https://jellyfin.org/docs/general/administration/hardware-acceleration/amd/