PeterSuh-Q3/tinycore-redpill #9: My nic:MT27710 Family [ConnectX-4 Lx] on model:918+ OS:7.2.1   is not working

yhysunwei, 2023-11-16:
NIC ConnectX-4 Lx is not working

PeterSuh-Q3, 2023-11-17:
dmesg | Can you give me the results of grep mlx5 ?

yhysunwei, 2023-11-17:
> Can you give me the results of dmesg | grep mlx5 ?

sunwei@test:/$ dmesg | grep mlx5
[   12.925245] mlx5_core 0000:03:00.0: enabling device (0000 -> 0002)
[   12.926115] mlx5_core 0000:03:00.0: firmware version: 14.32.1010
[   13.661500] mlx5_core 0000:03:00.0: 0000:03:00.0:handle_hca_cap:438:(pid 6016): log_max_qp value in current profile is 18, changing it to HCA capability limit (17)
[   13.672173] mlx5_core 0000:03:00.0: handle_hca_cap failed
[   14.059052] mlx5_core 0000:03:00.0: mlx5_load_one failed with error code -22
[   14.059582] mlx5_core: probe of 0000:03:00.0 failed with error -22

PeterSuh-Q3, 2023-12-07:
Sorry for the late reply.
Among the error logs you provided, the message "handle_hca_cap failed" caught my eye, so I looked it up on Google.
There were related improvement patches as follows.

https://patchwork.kernel.org/project/netdevbpf/patch/20210519060523.17875-15-saeed@kernel.org/

And, pci_irq.c containing this content exists only in the SA6400 model using kernel 5.

https://github.com/PeterSuh-Q3/arpl-modules/blob/main/src/5.x/drivers/net/ethernet/mellanox/mlx5/core/pci_irq.c

SA6400 supports graphics acceleration of Intel 11th generation or higher iGPU, enabling transcoding.
Device mapping is easy based on device-tree.
HBAs are also supported.
If it's a model that suits your purposes, we recommend giving the SA6400 a try.
This model is currently in a significant stabilization phase.

closed 2023-12-10 by PeterSuh-Q3