PeterSuh-Q3/tinycore-redpill #12: virtio gpu sa6400 13500 vgpu not work

njzydark, 2023-12-01:
I build the loader with pve. It can launch successful, but /dev/dri is empty

PeterSuh-Q3, 2023-12-01:
It will not work with vgpu.
I think I heard that the successful PVE case was connected by passing through the IGPU. Please refer to the community's success stories.

njzydark, 2023-12-05:
after add `i915.enable_guc=3` , it work: 

[image]

PeterSuh-Q3, 2023-12-07:
I did it as you suggested, but virtio vgpu was not accelerated.
These are my hardware settings and dmesg results.
I used Intel 4th generation Haswell.

[image]

[image]

No settings were added to the two files below.
/etc/modprobe.d/pve-blacklist.conf
/etc/modprobe.d/vfio-pci.conf

I tried adding the settings below to /etc/modules and removing them again.
It's same.
vfio
vfio_iommu_type1
vfio_pci
vfio_virqfd

What's your hardware setting like?

closed 2023-12-18 by PeterSuh-Q3