PeterSuh-Q3/tinycore-redpill #13: Tiny core friend won't boot without internet access

PrutzerNL, 2023-12-02:
I have build succesfully the loader. (lastest version)
I can start DSM if the internet connection is allowed on my firewall. But when I disable internet access to the nas and reboot it, tiny core friend is stuck and will not start DSM.

I changed friendautoupdate to false: “friendautoupd”: “false
I tried: “staticboot”: “false” but it is still not booting.
If I enable the internet connection it will boot succesfully.

How can I change this?
It is stuck on: 

Press <e> to enter a menu for edit USB/Sata Command Line
Press <j> to enter Junior mode

Pressing e or j is doing nothing.

PeterSuh-Q3, 2023-12-07:
You seem to be trying to use DSM within a local network without internet.

After 7 seconds, there is a process of downloading the Python library from the internet to display the QR.
Let's skip this process when there is no internet and make sure there are no errors.

It will be automatically updated when version 0.0.9m is upgraded.
Please reactivate friendaudoupdate and wait for new update.

PeterSuh-Q3, 2023-12-07:
However, when I disconnected the Internet and tested DS920+, I only failed to download the python library and succeeded in entering the DSM migration screen without Internet. (Because it waits for the pip script timeout, it takes quite a long time.)

Anyway, I will update 0.0.9m,
I don't think I'll be able to directly solve your problem.

[image]

[image]

PrutzerNL, 2023-12-07:
Thx for your reply and help!
I'm running a DVA 1622. I re-enabled auto update.
I tried today with todays version and waited for 10 minutes, but it is not booting.
I will try again with update 0.0.9m as soon as it is online.

Thanks a lot!

Screenshot:
[image]

PrutzerNL, 2023-12-08:
Thanks a lot! It is working now! :-)

PeterSuh-Q3, 2023-12-16:
Hi @PrutzerNL,

Thank you for your sponsor donation.

Sorry for the delay in thanking you.