PeterSuh-Q3/tinycore-redpill #15: mvsas9485 sub device id is 0x8635 and not 0x9485

stwad2, 2023-12-07:
Sorry, my controller sub device id is 0x8635 and not 0x9485. Can DSM not load properly solve this problem? thanks

Slot: 01:00.0
Class: Class 0104
Vendor: Vendor 1b4b
Device: Device 9485
SVendor: Unknown vendor 1043
SDevice: Device 8635
Rev: c3

Slot: 02:00.0
Class: Class 0104
Vendor: Vendor 1b4b
Device: Device 9485
SVendor: Unknown vendor 1043
SDevice: Device 8635

PeterSuh-Q3, 2023-12-07:
Did you try it with ddsml?
Can you provide dmesg ?

stwad2, 2023-12-08:
> Did you try it with ddsml? Can you provide dmesg ?
Yes, I have tried everything but couldn't recognize it correctly
[dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13612596/dmesg.txt)
[linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13612598/linuxrc.syno.log)

PeterSuh-Q3, 2023-12-10:
I don't see any mvsas related processes in dmesg.
Could you try eudev instead of ddsml?

stwad2, 2023-12-11:
> I don't see any mvsas related processes in dmesg. Could you try eudev instead of ddsml?
Thank you very much. I will try eudev and provide feedback again

stwad2, 2023-12-12:
Hello, sorry. Is this my network issue? I have tried many times, but all of them are the same mistake.
[image]

PeterSuh-Q3, 2023-12-13:
This problem occurs because there is a difference between the Internet time and the time of the PC you are trying to use.

PeterSuh-Q3, 2023-12-13:
It wasn't a problem with vision.
SSL(60) issue in the past: Checking the SSL certificate locally for 60 seconds
There was a -k option in the curl command to prevent
This option was left out by mistake.
I added the -k (--insecure) option again so that there is no certificate check.
Please try building again.

https://github.com/PeterSuh-Q3/redpill-load/commit/9322c46d5e35ac2096bcd92a74e84680bfff6977

stwad2, 2023-12-14:
> Thank you very much for building it normally, but 1b4b: 9485 still cannot be recognized correctly. I tried eudev but couldn't load it correctly. The current version I am using is DSM 6.1, and IG-88 helped me build the driver. It has explained the reason for the failure
[image]

PeterSuh-Q3, 2023-12-15:
I will also modify the mvsas module so that it can ignore the subdevice for this 0x9485 device.
Since this is an unfamiliar field, it may take some time and success is not guaranteed.

stwad2, 2023-12-15:
thank you. Looking forward to your success

PeterSuh-Q3, 2023-12-15:
The modified mvsas module has been applied.
Please rebuild the loader using version DS3622xs+ 7.2.1-69057.

stwad2, 2023-12-18:
Thank you very much. We are currently building tests and will provide feedback on the results later

stwad2, 2023-12-20:
Sorry! It took me so long to reply, and after testing many times, I still can't recognize the drive correctly. Do you need me to provide a log? Main board model ASUS P9A-I

PeterSuh-Q3, 2023-12-21:
I would appreciate it if you could give me the dmesg log.
If the log contains too much content, you can filter the results using the command below.
dmesg | grep mvsas

stwad2, 2023-12-21:
Hello, this is DSM 7.2.1-69057 dmesg log
[dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13737403/dmesg.txt)

PeterSuh-Q3, 2023-12-21:
I don't see any clues about the error.
Is there any difference between linuxrc.syno.log and the one you shared above?

stwad2, 2023-12-22:
Sorry, my computer skills are not very good. The file linuxrc.syno.log is on a USB drive or an HDD drive, and I couldn't find it in the DSM/var/log/directory.

Is this dmesg file, which is used normally in DSM6.2, helpful to you? Or mvsas files.
[6.2dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13747269/6.2dmesg.txt)

Sub device 9235?
[DSM6.2-lSpci.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13747270/DSM6.2-lSpci.txt)
[DSM7.2-LSpci.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13747271/DSM7.2-LSpci.txt)

stwad2, 2023-12-28:
I saw that the 9485 driver is correctly recognized in inulrc.syno.log, but cannot recognize the driver
[linuxrc.syno.log.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13785053/linuxrc.syno.log.txt)
[dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13785061/dmesg.txt)
Excuse me, do you still hope to use this motherboard normally?

PeterSuh-Q3, 2024-01-08:
I'm sorry. The answer was late.
I directly asked IG-88 if there was a problem with my patch coding.

https://xpenology.com/forum/topic/28321-driver-extension-jun-103b104b-for-dsm623-for-918-3615xs-3617xs/?do=findComment&comment=454794

PeterSuh-Q3, 2024-01-08:
Before hearing IG-88's response, I applied the following content exactly as IG-88's patch content.
{ PCI_VDEVICE(MARVELL_EXT, 0x9485), chip_9485 }, /* Marvell 9480/9485 (any vendor/model) */

Last time MARVELL was used instead of MARVELL_EXT.
The integrated module version has been changed from v1.64 to v1.65, and distribution has been completed for use in loader builds.

Could you please test mvsas one more time?

stwad2, 2024-01-09:
Thank you very much for your labor. I will compile and test it

stwad2, 2024-01-09:
> Before hearing IG-88's response, I applied the following content exactly as IG-88's patch content. { PCI_VDEVICE(MARVELL_EXT, 0x9485), chip_9485 }, /* Marvell 9480/9485 (any vendor/model) */
>
> Last [...]

Thank you very much for your efforts! 88se9485 has been recognized normally

Test DS3622 for normal recognition and installation of DSM

DS3617 installation of DMS to 52% indicates file damage

All can recognize the hard drive normally

DS3615 unable to load MVSAS driver

[DS3617 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874616/DS3617.linuxrc.syno.log)
[DS3617dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874617/DS3617dmesg.txt)
[DS3622 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874618/DS3622.linuxrc.syno.log)
[ds3622dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874619/ds3622dmesg.txt)
[image]

PeterSuh-Q3, 2024-01-09:
> > Before hearing IG-88's response, I applied the following content exactly as IG-88's patch content. { PCI_VDEVICE(MARVELL_EXT, 0x9485), chip_9485 }, /* Marvell 9480/9485 (any vendor/model) */
> > Last time MARVELL was used instead of MARVELL_EXT. The integrated module version has been changed from v1.64 to v1.65, and distribution has been completed for use in loader builds.
> > Could you please test mvsas one more time?
> 
> Thank you very much for your efforts! 88se9485 has been recognized normally
> 
> Test DS3622 for normal recognition and installation of DSM
> 
> DS3617 installation of DMS to 52% indicates file damage
> 
> All can recognize the hard drive normally
> 
> DS3615 unable to load MVSAS driver
> 
> [DS3617 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874616/DS3617.linuxrc.syno.log) [DS3617dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874617/DS3617dmesg.txt) [DS3622 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874618/DS3622.linuxrc.syno.log) [ds3622dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874619/ds3622dmesg.txt) ![mvsas](https://private-user-images.githubusercontent.com/151482726/295228503-574a85b7-e924-4d62-a9c7-11aa15bafa9e.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4MTIyNzYsIm5iZiI6MTcwNDgxMTk3NiwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1MjI4NTAzLTU3NGE4NWI3LWU5MjQtNGQ2Mi1hOWM3LTExYWExNWJhZmE5ZS5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwOVQxNDUyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMTczNTJhMGJlYWMwMTMwYmJmODk0Nzg3NTc2ZGQ2N2ZjNTc0M2RlNWQ5ODYzZGY0YzVlMTkyODUyMDcwMjUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.66Qqhqpp7pdQYZ9-p22ulxjHexPAw62u15Q1lMJyA6M)

It went really well.
I finally achieved it.
I will compile and apply version 7.1 in the same way.
It's rewarding to work hard together.
I need to give feedback to IG-88 as a success story.

stwad2, 2024-01-09:
> > > Before hearing IG-88's response, I applied the following content exactly as IG-88's patch content. { PCI_VDEVICE(MARVELL_EXT, 0x9485), chip_9485 }, /* Marvell 9480/9485 (any vendor/model) */
> > > Last time MARVELL was used instead of MARVELL_EXT. The integrated module version has been changed from v1.64 to v1.65, and distribution has been completed for use in loader builds.
> > > Could you please test mvsas one more time?
> > 
> > 
> > Thank you very much for your efforts! 88se9485 has been recognized normally
> > Test DS3622 for normal recognition and installation of DSM
> > DS3617 installation of DMS to 52% indicates file damage
> > All can recognize the hard drive normally
> > DS3615 unable to load MVSAS driver
> > [DS3617 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874616/DS3617.linuxrc.syno.log) [DS3617dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874617/DS3617dmesg.txt) [DS3622 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874618/DS3622.linuxrc.syno.log) [ds3622dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13874619/ds3622dmesg.txt) ![mvsas](https://private-user-images.githubusercontent.com/151482726/295228503-574a85b7-e924-4d62-a9c7-11aa15bafa9e.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4MTIyNzYsIm5iZiI6MTcwNDgxMTk3NiwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1MjI4NTAzLTU3NGE4NWI3LWU5MjQtNGQ2Mi1hOWM3LTExYWExNWJhZmE5ZS5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwOVQxNDUyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMTczNTJhMGJlYWMwMTMwYmJmODk0Nzg3NTc2ZGQ2N2ZjNTc0M2RlNWQ5ODYzZGY0YzVlMTkyODUyMDcwMjUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.66Qqhqpp7pdQYZ9-p22ulxjHexPAw62u15Q1lMJyA6M)
> 
> It went really well. I finally achieved it. I will compile and apply version 7.1 in the same way. It's rewarding to work hard together. I need to give feedback to IG-88 as a success story.
> 
> DS3615xs requires separate compilation. I will proceed with compiling DSM 7.1 version.

Thank you. Your efforts have revived my motherboard, and I will continue to test 3615xs. Thank you again!

PeterSuh-Q3, 2024-01-09:
In the case of DS3615xs, it seems impossible because there is no source code that can be compiled.
I'm sorry.

https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/3.x/drivers/scsi

PeterSuh-Q3, 2024-01-10:
I applied mvsas MARVELL 88se9485 to all platforms so 
that it works not only on DS3615xs (bromlow) but also on SA6400 (epyc7002).

https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/3.x/drivers/scsi/mvsas
https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/5.x/drivers/scsi/mvsas

https://github.com/PeterSuh-Q3/arpl-modules/releases/tag/v1.66

Can you give us feedback on the test results of both DS3615xs and SA6400 models?

stwad2, 2024-01-10:
> I applied mvsas MARVELL 88se9485 to all platforms so that it works not only on DS3615xs (bromlow) but also on SA6400 (epyc7002). [...]

Very willing

I tested 3615xs 7.1 and found that the mvsas driver failed to load automatically. Insmod mvsas was able to load successfully, and the system could recognize the driver
SA6400 is still being refactored, please provide feedback later
[image]
[3615ds-7.1 dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13884482/3615ds-7.1.dmesg.txt)

PeterSuh-Q3, 2024-01-10:
When using ddsml, mvsas has been added to be automatically detected in Junior.
Could you please rebuild and retest the DS3615xs loader?
If possible, please also provide Junior linuxrc.syno.log file.

https://github.com/PeterSuh-Q3/tcrp-modules/commit/c919d29dbda323318d5fa9c090fe3d392aff2462

stwad2, 2024-01-10:
> When using ddsml, mvsas has been added to be automatically detected in Junior. Could you please rebuild and retest the DS3615xs loader? If possible, please also provide Junior linuxrc.syno.log file.
> 
> [PeterSuh-Q3/tcrp-modules@c919d29](https://github.com/PeterSuh-Q3/tcrp-modules/commit/c919d29dbda323318d5fa9c090fe3d392aff2462)

OK
I am rebuilding

stwad2, 2024-01-10:
> I applied mvsas MARVELL 88se9485 to all platforms so that it works not only on DS3615xs (bromlow) but also on SA6400 (epyc7002). [...]
After installing DSM on SA6400 and starting normally, the drive cannot be recognized,
The driver program is loading normally
[SA6400 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885585/SA6400.linuxrc.syno.log)
[SA6400.dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885586/SA6400.dmesg.txt)
[SA6400.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885587/SA6400.txt)
[image]
[image]

PeterSuh-Q3, 2024-01-10:
> > I applied mvsas MARVELL 88se9485 to all platforms so that it works not only on DS3615xs (bromlow) but also on SA6400 (epyc7002).
> > https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/3.x/drivers/scsi/mvsas https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/5.x/drivers/scsi/mvsas
> > https://github.com/PeterSuh-Q3/arpl-modules/releases/tag/v1.66
> > Can you give us feedback on the test results of both DS3615xs and SA6400 models?
> > After installing DSM on SA6400 and starting normally, the drive cannot be recognized,
> > The driver program is loading normally
> > [SA6400 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885585/SA6400.linuxrc.syno.log)
> > [SA6400.dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885586/SA6400.dmesg.txt)
> > [SA6400.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885587/SA6400.txt)
> > ![SA6400](https://private-user-images.githubusercontent.com/151482726/295514164-d38457b8-ab83-461c-b5e3-73a14d9ded96.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4ODE4MjksIm5iZiI6MTcwNDg4MTUyOSwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1NTE0MTY0LWQzODQ1N2I4LWFiODMtNDYxYy1iNWUzLTczYTE0ZDlkZWQ5Ni5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMFQxMDEyMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYzdlYzBiMzdlNWE1Y2Q3MTE4ZTY0NjIxNjQ1MjkzYjRiMGVmMGNlYTQ2YzVlY2Y0ZmYwOWQ3MTU1ODlmNzkwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D9ATcd5dJRfKDoP3QPl8aKnJIYNiQRnDQyulr64QlCQ)
> > ![SA6400mvsas](https://private-user-images.githubusercontent.com/151482726/295514166-3266e75e-0e6f-44d8-b099-2dd857db5897.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4ODE4MjksIm5iZiI6MTcwNDg4MTUyOSwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1NTE0MTY2LTMyNjZlNzVlLTBlNmYtNDRkOC1iMDk5LTJkZDg1N2RiNTg5Ny5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMFQxMDEyMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTBlY2U5YWU0NmJiMzFjOTVmZDc0Yjk4OGRjZjYyOGU0M2UxYjM4NzdlMGJkMTcwNmU4NTk3NDZhMjg5OWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-KwSPOKIElYwdDWGBtVSs-WMKLTOsJPEyRvCXeUb6QE)

The mvsas module itself for sa6400 is probably working well.
Since this model is Device-Tree based, disk mapping through disks dddon is required separately.
I've been spending a lot of time recently mapping for mpt3sas.
Since I actually own a Dell Perc H310, testing was easy.
I don't think I can ask you any more because it seems difficult for you to keep changing the model and testing it.
Lastly, can you give me the results of the command below in junior mode?

ll /sys/block

Based on these results I will determine if any disks addon modifications are necessary.

stwad2, 2024-01-11:
> The mvsas module itself for sa6400 is probably working well. Since this model is Device-Tree based, disk mapping through disks dddon is required separately. [...]

Sorry for not responding to the test in a timely manner due to work and time difference reasons 

I encountered an error during the process of building 3615xs. Do I need to reformat the USB drive?
[image]

PeterSuh-Q3, 2024-01-11:
> > > > I applied mvsas MARVELL 88se9485 to all platforms so that it works not only on DS3615xs (bromlow) but also on SA6400 (epyc7002).
> > > > https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/3.x/drivers/scsi/mvsas https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/5.x/drivers/scsi/mvsas
> > > > https://github.com/PeterSuh-Q3/arpl-modules/releases/tag/v1.66
> > > > Can you give us feedback on the test results of both DS3615xs and SA6400 models?
> > > > After installing DSM on SA6400 and starting normally, the drive cannot be recognized,
> > > > The driver program is loading normally
> > > > [SA6400 linuxrc.syno.log](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885585/SA6400.linuxrc.syno.log)
> > > > [SA6400.dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885586/SA6400.dmesg.txt)
> > > > [SA6400.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13885587/SA6400.txt)
> > > > ![SA6400](https://private-user-images.githubusercontent.com/151482726/295514164-d38457b8-ab83-461c-b5e3-73a14d9ded96.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4ODE4MjksIm5iZiI6MTcwNDg4MTUyOSwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1NTE0MTY0LWQzODQ1N2I4LWFiODMtNDYxYy1iNWUzLTczYTE0ZDlkZWQ5Ni5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMFQxMDEyMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYzdlYzBiMzdlNWE1Y2Q3MTE4ZTY0NjIxNjQ1MjkzYjRiMGVmMGNlYTQ2YzVlY2Y0ZmYwOWQ3MTU1ODlmNzkwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D9ATcd5dJRfKDoP3QPl8aKnJIYNiQRnDQyulr64QlCQ)
> > > > ![SA6400mvsas](https://private-user-images.githubusercontent.com/151482726/295514166-3266e75e-0e6f-44d8-b099-2dd857db5897.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4ODE4MjksIm5iZiI6MTcwNDg4MTUyOSwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1NTE0MTY2LTMyNjZlNzVlLTBlNmYtNDRkOC1iMDk5LTJkZDg1N2RiNTg5Ny5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMFQxMDEyMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTBlY2U5YWU0NmJiMzFjOTVmZDc0Yjk4OGRjZjYyOGU0M2UxYjM4NzdlMGJkMTcwNmU4NTk3NDZhMjg5OWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-KwSPOKIElYwdDWGBtVSs-WMKLTOsJPEyRvCXeUb6QE)
> > 
> > 
> > The mvsas module itself for sa6400 is probably working well. Since this model is Device-Tree based, disk mapping through disks dddon is required separately.
> > https://github.com/PeterSuh-Q3/tcrp-addons/blob/main/disks/releases/install.sh
> > I've been spending a lot of time recently mapping for mpt3sas. Since I actually own a Dell Perc H310, testing was easy. I don't think I can ask you any more because it seems difficult for you to keep changing the model and testing it. Lastly, can you give me the results of the command below in junior mode?
> > ll /sys/block
> > Based on these results I will determine if any disks addon modifications are necessary.
> 
> Sorry for not responding to the test in a timely manner due to work and time difference reasons
> 
> I encountered an error during the process of building 3615xs. Do I need to reformat the USB drive? ![212](https://private-user-images.githubusercontent.com/151482726/295749712-07f6a9d6-8089-4104-bd87-6342cc34441e.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5MzQ1NzksIm5iZiI6MTcwNDkzNDI3OSwicGF0aCI6Ii8xNTE0ODI3MjYvMjk1NzQ5NzEyLTA3ZjZhOWQ2LTgwODktNDEwNC1iZDg3LTYzNDJjYzM0NDQxZS5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQwMDUxMTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNzc2NzlmNzUzNGQyMzg3NjI2OTdiZTkzYTY4YjE0YmIzZmQ3ZDg2Yjg0MDA1YTFiZDgwOTZkMmY3OTY4ZDhhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tPgTzl2684dTBLUghqn3UcdOvElNs2UWyR91dyFhSSg)

I will check again and edit the SHA256 contents of the file.

PeterSuh-Q3, 2024-01-11:
I'm sorry.

I found the cause of the error and took action.

Please try again.

stwad2, 2024-01-11:
> I'm sorry.
> 
> I found the cause of the error and took action.
> 
> Please try again.
OK, let me try. Sorry, due to network firewall issues, we often need a proxy to access.

stwad2, 2024-01-11:
Hi, there are two more positions reporting errors
[image]
[image]

PeterSuh-Q3, 2024-01-11:
Would you like to try the build one more time?
/home/tc/redpill-load is causing problems
After the directory has been completely cleaned up and deleted
The regeneration process should resolve the error.

stwad2, 2024-01-11:
Sorry for taking so long to provide feedback. The driver recognized it normally. I will install DSM and check the results
[linuxrc.syno.log.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13897070/linuxrc.syno.log.txt)
[ll sysblock.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13897072/ll.sysblock.txt)
[image]

PeterSuh-Q3, 2024-01-11:
Thank you for your feedback.

Are all results from DS3615xs?
There is no SA6400, right?

In the results of Junior's linuxrc.syno.log you provided, it was confirmed that the module was loaded properly as shown below.

Found RAID bus Controller: pciid 1b4bd00009485 Required Extension: Searching for matching extension for mvsas
Found RAID bus Controller: pciid 1b4bd00009485 Required Extension: Searching for matching extension for mvsas
Module mvsas loaded succesfully

However, after that, automatic switching loading must be attempted once more at the time of DSM on-loading, but it does not seem to be working properly.
It seems difficult to find the cause easily.
If manually copied mvsas.ko does not load automatically,
I think it would be possible to run insmod on the scheduler using the root account.
Since the disk is loaded too late, it seems to be fatal to the RAID volume configuration.

If you do not need to use DS3615xs, we recommend switching to DS3617xs or DS3622xs+, which use DSM kernel 4.

If I come up with a good idea later, I'll try again.
Thank you again for your feedback.

stwad2, 2024-01-11:
> Thank you for your feedback.
> 
> Are all results from DS3615xs? There is no SA6400, right?
> 
> In the results of Junior's linuxrc.syno.log you provided, it was confirmed that the module was loaded properly as shown below.
> 
> Found RAID bus Controller: pciid 1b4bd00009485 Required Extension: Searching for matching extension for mvsas Found RAID bus Controller: pciid 1b4bd00009485 Required Extension: Searching for matching extension for mvsas Module mvsas loaded succesfully
> 
> However, after that, automatic switching loading must be attempted once more at the time of DSM on-loading, but it does not seem to be working properly. It seems difficult to find the cause easily. If manually copied mvsas.ko does not load automatically, I think it would be possible to run insmod on the scheduler using the root account. Since the disk is loaded too late, it seems to be fatal to the RAID volume configuration.
> 
> If you do not need to use DS3615xs, I recommend switching to DS3617xs or DS3622xs+, which use DSM kernel 4.
> 
> If I come up with a good idea later, I'll try again. Thank you again for your feedback.

Due to time constraints, SA6400 needs to be tested later. We will provide you with feedback later

Thank you for your efforts

stwad2, 2024-01-12:
> Thank you for your feedback.
>
> Are all results from DS3615xs? There is no SA6400, right?
>
> In the results of Junior's linuxrc.syno.log you provided, it was confirmed that the module was loaded [...]

I'm sorry for taking so long to provide feedback. SA6400mvsas driver everything is normal
[image]
[image]
[image]

PeterSuh-Q3, 2024-01-15:
> > Thank you for your feedback.
> > Are all results from DS3615xs? There is no SA6400, right?
> > In the results of Junior's linuxrc.syno.log you provided, it was confirmed that the module was loaded properly as shown below.
> > Found RAID bus Controller: pciid 1b4bd00009485 Required Extension: Searching for matching extension for mvsas Found RAID bus Controller: pciid 1b4bd00009485 Required Extension: Searching for matching extension for mvsas Module mvsas loaded succesfully
> > However, after that, automatic switching loading must be attempted once more at the time of DSM on-loading, but it does not seem to be working properly. It seems difficult to find the cause easily. If manually copied mvsas.ko does not load automatically, I think it would be possible to run insmod on the scheduler using the root account. Since the disk is loaded too late, it seems to be fatal to the RAID volume configuration.
> > If you do not need to use DS3615xs, I recommend switching to DS3617xs or DS3622xs+, which use DSM kernel 4.
> > If I come up with a good idea later, I'll try again. Thank you again for your feedback.
> 
> I'm sorry for taking so long to provide feedback. SA6400mvsas driver everything is normal ![SA6400](https://private-user-images.githubusercontent.com/151482726/296105790-8a059fa7-387f-4992-9cb4-0f00167db725.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyODM2MjMsIm5iZiI6MTcwNTI4MzMyMywicGF0aCI6Ii8xNTE0ODI3MjYvMjk2MTA1NzkwLThhMDU5ZmE3LTM4N2YtNDk5Mi05Y2I0LTBmMDAxNjdkYjcyNS5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQwMTQ4NDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMDQ4YmEyMTY1YmYwNGIwOTM4ZjA2YmNhMzhiYWMyZjJhNTg5Y2NkMGVkNGQ0YzdjNjY2MGYxZWFlMWI1MGVmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kzWH0YjiNN_WAJ8P4ntThY29wNXhqJ6LGrlMDdYeHFs) ![SA6400-1](https://private-user-images.githubusercontent.com/151482726/296105793-96cc0488-a8fd-435e-8ad7-9662fc114fc3.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyODM2MjMsIm5iZiI6MTcwNTI4MzMyMywicGF0aCI6Ii8xNTE0ODI3MjYvMjk2MTA1NzkzLTk2Y2MwNDg4LWE4ZmQtNDM1ZS04YWQ3LTk2NjJmYzExNGZjMy5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQwMTQ4NDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZDRhM2E2MDI5N2VhZTc5OTI3ZDYzZjRlNjBlNGY5Njk3NGM2YTZmMTcxN2Y5NGU0ZjE5MGY1Mjg5NGViZjY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Cdtmbh3LXP2YguX-fcv-9TeUJZXN2t2lXC-6YG28xFg) ![SA6400-2](https://private-user-images.githubusercontent.com/151482726/296105795-e1650757-04cc-405f-a13c-b48f981e8658.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyODM2MjMsIm5iZiI6MTcwNTI4MzMyMywicGF0aCI6Ii8xNTE0ODI3MjYvMjk2MTA1Nzk1LWUxNjUwNzU3LTA0Y2MtNDA1Zi1hMTNjLWI0OGY5ODFlODY1OC5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQwMTQ4NDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ODg5NDA2MmMyMzkzZTRiZWFiMzViNTlkNmIyZDY1ZTc2NWFmMzQwZWI1MzMxNDEzMmQ1ZDVkM2Y0ZTg3YjFkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FKwAJEhnFxDZsgBaHsLqXgfgEASid-Z1OBxS_eCLZfg)

Thank you for your feedback.
SA6400 is designed to operate only with EUDEV.

There seems to be no problem with the module itself, but there is still an issue where 4 disks cannot be activated in later modes, including Junior.
I mainly tested with mpt3sas, but I will purchase another SAS controller and conduct additional tests.

If necessary, we will request another test later.
We will close the issue.
Thank you for everything.

closed 2024-01-15 by PeterSuh-Q3

PeterSuh-Q3, 2024-02-18:
Today, in mshell, we made it almost identical to the mechanism of rr so that the HBA operates stably.
All HBAs that did not work until now will work.
If possible, please test it.