PeterSuh-Q3/tinycore-redpill #16: Build with mpt3sas rather then automaticly fond megaraid_sas module

Lostish, 2023-12-09:
Hi

I want to override megaraid_sas to mpt3sas module.

Some background.
Im using ESXi 8.0u2 with xeon brodwell cpus.
Im choosing the broadwell platform because of the underlying host cpu matches.
Probably unnessecary but its nice to my eyes.
Passthrough LSI 3008 card to the host.

I have tried to get LSI 3008 card working with compiling xpenology with redpill.
The automatic prober wants to use the megaraid_sas module.
But that does not work at all, i need to

rmmod megaraid_sas
insmod megaraid_sas

to get the system to want to install.
And then when rebooted im meet with no disks avalible.
And im in a loop.

SO:

Im trying to add the mpt3sas module and the system will not compile.
./redpill-load/ext-manager.sh add https://raw.githubusercontent.com/PeterSuh-Q3/rp-ext/master/mpt3sas/rpext-index.json
OR
./rploader.sh ext rs3618xs-7.1.1-42962 add https://raw.githubusercontent.com/PeterSuh-Q3/rp-ext/master/mpt3sas/rpext-index.json
It is the same

I see in the repo that there is a module for 4.4.180+ kernel, so anything above DSM-7.1.1 uses a newer kernel, i understand that.
So, trying to add and build 7.1.1 ( i have tried almost every system in the list.... ) the mpt3sas modules spits out the same error.

Now i tested PeterSuh's extentions but it is the same as pocopico, i get the same errors:

```
tc@box:~$ ./rploader.sh build rs3618xs-7.1.1-42962 static
platform_selected={
  "id": "rs3618xs-7.1.1-42962",
  "platform_name": "broadwell",
  "downloads": {
    "os": {
      "sha256": "2ccd7dc0c33cb133476bce8cdd29db6a164669c0e0caf6fb2daad17202525ec5"
    }
  },
  "add_extensions": [
    {
      "all-modules": "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-modules/master/all-modules/rpext-index.json",
      "storagepanel": "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-addons/master/storagepanel/rpext-index.json"
    }
  ]
}
Rploader Version  : 1.0.0.5
Extensions        :   all-modules
  storagepanel
Extensions URL    : "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-modules/master/all-modules/rpext-index.json",
"https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-addons/master/storagepanel/rpext-index.json"
TARGET_PLATFORM   : rs3618xs
TARGET_VERSION    : 7.1.1
TARGET_REVISION   : 42962
KERNEL_MAJOR      : 4
MODULE_ALIAS_FILE : modules.alias.4.json
SYNOMODEL         : rs3618xs_42962
MODEL             : RS3618xs
KERNEL VERSION    : 4.4.180
Local Cache Folder : /mnt/sda3/auxfiles
DATE Internet     : 09122023 Local : 09122023
Checking Internet Access -> OK
KERNEL VERSION of getredpillko() is 4.4.180, DSMVER is 7.1
Downloading broadwell 4.4.180+ redpill.ko ...
TAG is 23.6.0
'/tmp/rp-broadwell-4.4.180-prod.ko' -> '/home/tc/custom-module/redpill.ko'
static

No extra build option or static specified, using default <static>
Using static compiled redpill extension
'/home/tc/custom-module/redpill.ko' -> '/home/tc/redpill-load/ext/rp-lkm/redpill-linux-v4.4.180+.ko'
Got redpill-linux-v4.4.180+.ko
Testing modules.alias.4.json -> File OK
------------------------------------------------------------------------------------------------
It looks that you will need the following modules :

Ethernet Interface : Required Extension :
Ethernet Interface : Required Extension : vmxnet3
RAID bus Controller : Required Extension : megaraid_sas
LSI Logic / Symbios Logic MegaRAID SAS-3 3008 [Fury]  (rev 02)
------------------------------------------------------------------------------------------------
```

```
[#] Updating broadwell_44180 platforms extensions...
[-] The extension PeterSuh-Q3.mpt3sas was found. However, the extension index has no recipe for broadwell_44180 platform. It may not be
[-] supported on that platform, or author didn't updated it for that platform yet. You can try running
[-] "ext-manager.sh update" to refresh indexes for all extensions manually. Below are the currently known information about
[-] the extension stored locally:
[#] ========================================== PeterSuh-Q3.mpt3sas ==========================================
[#] Extension name: mpt3sas
[#] Description: Adds LSI MPT Fusion SAS 3.0 Device Driver Support
[#] To get help visit: <todo>
[#] Extension preparer/packer: https://github.com/PeterSuh-Q3/rp-ext/tree/main/mpt3sas
[#] Software author: https://github.com/PeterSuh-Q3
[#] Update URL: https://raw.githubusercontent.com/PeterSuh-Q3/rp-ext/master/mpt3sas/rpext-index.json
[#] Platforms supported: ds3617xs_42218 dva3221_42962 ds1019p_42218 ds3622xsp_42962 rs3618xs_42962 dva3221_42218 ds3622xsp_42218 rs3413xsp_42218 rs4021xsp_42962 zendofmodel ds3615xs_42962 ds3617xs_42962 rs3618xs_42218 rs4021xsp_42218 dva3219_42962 dva3219_42218 ds1621xsp_42962 ds918p_42962 ds918p_42218 ds1019p_42962 ds1621xsp_42218 rs3413xsp_42962 ds3615xs_42218
[#] =======================================================================================

[-] Failed to update recipe for PeterSuh-Q3.mpt3sas extension for platform broadwell_44180. The script will terminate as you do not
[-] have previously downloaded recipe which can be used if download fails. Try again later. If problem
[-] persists contact the extension packer for support (displayed below)
[#] ========================================== PeterSuh-Q3.mpt3sas ==========================================
[#] Extension name: mpt3sas
[#] Description: Adds LSI MPT Fusion SAS 3.0 Device Driver Support
[#] To get help visit: <todo>
[#] Extension preparer/packer: https://github.com/PeterSuh-Q3/rp-ext/tree/main/mpt3sas
[#] Software author: https://github.com/PeterSuh-Q3
[#] Update URL: https://raw.githubusercontent.com/PeterSuh-Q3/rp-ext/master/mpt3sas/rpext-index.json
[#] Platforms supported: ds3617xs_42218 dva3221_42962 ds1019p_42218 ds3622xsp_42962 rs3618xs_42962 dva3221_42218 ds3622xsp_42218 rs3413xsp_42218 rs4021xsp_42962 zendofmodel ds3615xs_42962 ds3617xs_42962 rs3618xs_42218 rs4021xsp_42218 dva3219_42962 dva3219_42218 ds1621xsp_42962 ds918p_42962 ds918p_42218 ds1019p_42962 ds1621xsp_42218 rs3413xsp_42962 ds3615xs_42218
[#] =======================================================================================
```

I have tried to figure out how everyting incorperate with each other but im giving up :)
As the broadwell_44180 specefiy the architecture and 44180 is the kernel.

Everything matches with the info in the mpt3sas repo.

How do i link/add broadwell_44180 with pocopico.mpt3sas/PeterSuh-Q3.mpt3sas to get it to build?
( I got this to work one time after like 6h testing at 4AM but 100 of builds and fiddeling around, i cant replicate it... :( So i need help)

PeterSuh-Q3, 2023-12-10:
> Hi
> 
> I want to override megaraid_sas to mpt3sas module.
> 
> Some background. Im using ESXi 8.0u2 with xeon brodwell cpus. Im choosing the broadwell platform because of the underlying host cpu matches. Probably unnessecary but its nice to my eyes. Passthrough LSI 3008 card to the host. ( In IR mode, i can flash it to IT mode but that is tedious :( )
> 
> I have tried to get LSI 3008 card working with compiling xpenology with redpill. The automatic prober wants to use the megaraid_sas module. But that does not work at all, i need to
> 
> rmmod megaraid_sas insmod megaraid_sas
> 
> to get the system to want to install. And then when rebooted im meet with no disks avalible. And im in a loop.
> 
> SO:
> 
> Im trying to add the mpt3sas module and the system will not compile. ./redpill-load/ext-manager.sh add https://raw.githubusercontent.com/PeterSuh-Q3/rp-ext/master/mpt3sas/rpext-index.json OR ./rploader.sh ext rs3618xs-7.1.1-42962 add https://raw.githubusercontent.com/PeterSuh-Q3/rp-ext/master/mpt3sas/rpext-index.json It is the same
> 
> I see in the repo that there is a module for 4.4.180+ kernel, so anything above DSM-7.1.1 uses a newer kernel, i understand that. So, trying to add and build 7.1.1 ( i have tried almost every system in the list.... ) the mpt3sas modules spits out the same error.
> 
> Now i tested PeterSuh's extentions but it is the same as pocopico, i get the same errors:
> 
> ```
> tc@box:~$ ./rploader.sh build rs3618xs-7.1.1-42962 static
> platform_selected={
>   "id": "rs3618xs-7.1.1-42962",
>   "platform_name": "broadwell",
>   "downloads": {
>     "os": {
>       "sha256": "2ccd7dc0c33cb133476bce8cdd29db6a164669c0e0caf6fb2daad17202525ec5"
>     }
>   },
>   "add_extensions": [
>     {
>       "all-modules": "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-modules/master/all-modules/rpext-index.json",
>       "storagepanel": "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-addons/master/storagepanel/rpext-index.json"
>     }
>   ]
> }
> Rploader Version  : 1.0.0.5
> Extensions        :   all-modules
>   storagepanel
> Extensions URL    : "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-modules/master/all-modules/rpext-index.json",
> "https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-addons/master/storagepanel/rpext-index.json"
> TARGET_PLATFORM   : rs3618xs
> TARGET_VERSION    : 7.1.1
> TARGET_REVISION   : 42962
> KERNEL_MAJOR      : 4
> MODULE_ALIAS_FILE : modules.alias.4.json
> SYNOMODEL         : rs3618xs_42962
> MODEL             : RS3618xs
> KERNEL VERSION    : 4.4.180
> Local Cache Folder : /mnt/sda3/auxfiles
> DATE Internet     : 09122023 Local : 09122023
> Checking Internet Access -> OK
> KERNEL VERSION of getredpillko() is 4.4.180, DSMVER is 7.1
> Downloading broadwell 4.4.180+ redpill.ko ...
> TAG is 23.6.0
> '/tmp/rp-broadwell-4.4.180-prod.ko' -> '/home/tc/custom-module/redpill.ko'
> static
> 
> No extra build option or static specified, using default <static>
> Using static compiled redpill extension
> '/home/tc/custom-module/redpill.ko' -> '/home/tc/redpill-load/ext/rp-lkm/redpill-linux-v4.4.180+.ko'
> Got redpill-linux-v4.4.180+.ko
> Testing modules.alias.4.json -> File OK
> ------------------------------------------------------------------------------------------------
> It looks that you will need the following modules :
> 
> 
> Ethernet Interface : Required Extension :
> Ethernet Interface : Required Extension : vmxnet3
[... 46 lines cut ...]
> Everything matches with the info in the mpt3sas repo.
> 
> How do i link/add broadwell_44180 with pocopico.mpt3sas/PeterSuh-Q3.mpt3sas to get it to build? ( I got this to work one time after like 6h testing at 4AM but 100 of builds and fiddeling around, i cant replicate it... :( So i need help)

It is correct that LSI SAS3008 Fury must use the megaraid_sas module.
I think @pocopico would have said the same thing.

https://linux-hardware.org/index.php?id=pci:1000-005f-1000-9343

Perhaps megaraid_sas used in TCRP-mshell has not been verified and may not be stable.

rr is currently unavailable, but the final image is being shared in the community and a loader can be built.
Please try megaaid_sas with rr.
If you tell me about stable operation, I will apply it to mshell as well.

mshell now has the ability to add individual modules discontinued.
All modules are already available in the form of integrated packs like rr.

The ramdisk in the loader already contains mpt3sas.ko,
The module is injected only when a matching device is found.
If you want to try switching to the mpt3sas module that is already included through TTYD and junior, you can try as follows. But I don't think we can expect positive results.

cd /lib64/modules
rmmod megaaid_sas.ko
insmod mpt3sas.ko

closed 2023-12-18 by PeterSuh-Q3

PeterSuh-Q3, 2024-02-18:
Today, in mshell, we made it almost identical to the mechanism of rr so that the HBA operates stably.
All HBAs that did not work until now will work.
If possible, please test it.