PeterSuh-Q3/tinycore-redpill #18: iGPU support for N6005

xavier-thomas, 2023-12-15:
Hi Peter,

First of all, thank you very much for all the great work you've done.
I'm trying to use a Intel N6005 board to run a plex host and I can't get HW Transcoding to work.

The `/dev/dri` device is not found.

Is this a missing driver issue ? and if so is there a way to package the drivers and enable the iGPU ?

everything else about the system is super stable and works very well, so it would be superb if I can get the iGPU enabled.

PeterSuh-Q3, 2023-12-16:
Hi Thomas.

Thank you for your sponsor donation.

In N6005 Jasper Lake, existing Synology models DS918+, DS920+, etc. cannot use iGPU acceleration for transcoding.

SA6400, which uses the newly developed and stabilized kernel 5, is the only solution.

The advantage of this model is the same Device-Tree as DS920+, which allows for easy device automatic mapping.
And, it even supports HBAs.

As you know, plex pass is required for HW transcoding.
However, SA6400, with Kernel 5's fast transcoding processing ability, has excellent processing ability even if it is SW transcoding.
Compared to existing models, the CPU load is relatively very low.

And, SA6400 does not need SYNO genuine SN or AME patches, which were required for existing models.

After installing DSM, check whether the /dev/dri directory exists and
dmesg | All you have to do is check whether the i915 module has been loaded without any problems using grep i915, etc.

xavier-thomas, 2023-12-16:
Hi @PeterSuh-Q3 thank you for the reply.

I tried to do as you suggested and followed the instructions to set up my device as a SA6400. 

I am able to use tinycore to build the Friend and reboot. However, once the friend loads, it gets an ip, goes into the boot phase and then the screen then goes completely blank but the box is still running. I tried waiting for around 15 - 20 minutes and trying `find.synology.com` but the device is not detected at all. 

It looks like the friend bootloader crashed.
Any idea what the problem is ?

I'm using a Topton n6005 motherboard.
When I tried setting it up as a DS920 earlier, it was working perfectly.

PeterSuh-Q3, 2023-12-17:
I searched for topton N6005 on AliExpress.
Is the product below correct?

https://www.aliexpress.com/item/1005004761498719.html

The NIC chipset shown in the product specifications is Intel i226-v.

This chipset has had issues with various OSes.

For Windows, Intel proactively supplemented drivers, but

The case of Linux is a little different.

You must verify stability after patching all i226-v modules corresponding to each kernel.

In the case of TCRP-mshell, stabilization testing is in progress.

In the case of TCRP-mshell, three kernels are used.

Tinycore Linux kernel, which corresponds to the first menu-based screen,
2nd Friend kernel,
3rd DSM kernel

All igc modules running the i226-v chipset in these three kernels must operate normally.

This test is scheduled for today with a Korean user using i226-v.
We are going to test it on DS918+ and SA6400 models.

There was an issue with the first Tinycore Linux version 12.x kernel, which prevented i226-v from operating.

It seems like you are in a slightly different situation in that you passed this part.

Also, I thought it would be a good idea to test it myself with the i226-v chipset, so I purchased the same NIC from AliExpress and it is currently being delivered quickly.

This is because there are limits to requesting testing through users.

We will conduct a test with Korean users scheduled for today and notify you when stability is confirmed.

PeterSuh-Q3, 2023-12-17:
A while ago, the friend kernel was updated from 0.1.0 to 0.1.0a.

[image]

Now you can visually see all four IP addresses.
With a visible IP address other than find.synology.com

Try a ping test first in advance.
IP Address: 5000 Please test direct connection.
Please feed back the results.

xavier-thomas, 2023-12-17:
Yes, that's the correct motherboard.

I'll try the process again and try recording / screen capture for you so it will help.

xavier-thomas, 2023-12-17:
Good news. I ran through the install process again and somehow it managed to complete the installation.
`/dev/dri` exists and i've passed it through to plex and tested that HW transcoding works.

[image]
[image]

PeterSuh-Q3, 2023-12-18:
Thank you for your feedback.
Transcoding works as expected.

However, yesterday, in the feedback from a user in Korea using a similar AliExpress sales model as yours, a problem was reported in which three NICs out of four i226-v (igc modules) could not be woken up again after being converted to EEE (Energy Efficient Ethernet) state. received.

Among the igc modules I compiled and the rr version of the modules, I am looking for modules for which stability has been established.
I think it would be a good idea to check this part one more time.
Currently, the igc module compiled directly by mshell is used in v1.64.
https://github.com/PeterSuh-Q3/arpl-modules/releases/tag/v1.64

everwisher, 2023-12-24:
I'm curious if AlderLake processors like N100 has a chance of getting HW acceleration with the SA6400 model profile?

PeterSuh-Q3, 2023-12-25:
> I'm curious if AlderLake processors like N100 has a chance of getting HW acceleration with the SA6400 model profile?

Yes, it is possible.
Recently, @MoetaYuko developed the i915 module for stable and fast transcoding HW acceleration in SA6400.
The first environment he tested was N100, Jellyfin.

everwisher, 2023-12-27:
> > I'm curious if AlderLake processors like N100 has a chance of getting HW acceleration with the SA6400 model profile?
> 
> Yes, it is possible. Recently, @MoetaYuko developed the i915 module for stable and fast transcoding HW acceleration in SA6400. The first environment he tested was N100, Jellyfin.

Thank you for letting me know his work. I've succeeded in getting his driver working on a physical xpenology.

closed 2024-01-07 by PeterSuh-Q3