PeterSuh-Q3/tinycore-redpill #20: no internet after booting up

dufka66, 2023-12-20:
Hi PeterSuh-Q3,

Firstly i really appreciate the work thank you for that!

I have an issue with the loader. Once it loads i see a purple line rops_handle_POLLIN: DELADDR and then the UTP link from the back goes off and i dont have any connection to the IP anymore. Attached a screenshot. Thanks again for the support.

Kind regards
[image]

PeterSuh-Q3, 2023-12-21:
It has not yet been revealed that the "rops_handle_POLLIN: DELADDR" message is directly related to the UTP Link being turned off. In most cases, the NIC operates normally despite the above message.

If this is your first time using Redpill, your NIC may have some special features.

Can you go back to the TCRP Loader Build menu, download all the files in the directory below, and attach them to the issue here?
Let us analyze the log.
/mnt/sda1/logs/jr

[image]

dufka66, 2023-12-21:
Hi Peter,

Thnx for the quick reply. Problem is i dont have a folder logs where i can troubleshoot. I have been running redpill for some time now but on a proxmox VM but now i want to move to a baremetal so i can use all the resources for XPE. Here is a screenshot of my /mnt/sda1 folder 

[image]

[image]

Kind regards

PeterSuh-Q3, 2023-12-21:
Does Nvme also have TCRP bootloader content?
I don't think I've ever written a script to mount Nvme.
The two partitions of nvme appear to be mounted.
What is this nvme for and what does it contain?

PeterSuh-Q3, 2023-12-21:
In very rare cases, nvme may prevent the Friend kernel from booting.
In the case of the ASUS IOT H310i-IM-A R2.0 motherboard I use, M.2 Nvme cannot be used as storage.
This is a case where an infinite error occurs in the kernel and the kernel panics.
I'm not sure what NVME is for, but would you try booting with it removed?

dufka66, 2023-12-21:
Tested withouth the nvme and same behaviour ethernet port lamp goes out after the comment and no log folder once i boot back to build. Any other suggestions its driving me nuts.

PeterSuh-Q3, 2023-12-21:
Without dmesg or linuxrc.syno.log, it is difficult to analyze the cause.

If the nic does not work, the COM1 serial port can take over this role, 
but it is difficult for ordinary users to have this environment.

So, the last suggestion seems to be to install another Intel-based NIC separately instead of the r8169 Realtek NIC.

closed 2023-12-24 by PeterSuh-Q3