PeterSuh-Q3/tinycore-redpill #23: DSM 무한재설치 문의드립니다

Fentanest, 2024-01-06:
서버포럼에도 글을 쓰긴 했는데...
여기다가도 올려봅니다.

로그가 길어 별도 첨부파일로 두 개 올리겠습니다.

[linuxrc.syno.log .txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13850062/linuxrc.syno.log.txt)
[messages.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13850063/messages.txt)

설치에 사용한 제품은 인텔 1U서버이고

 

서버 제품명은 R1304RPOSHBN

(https://www.intel.co.kr/content/www/kr/ko/products/sku/71389/intel-server-system-r1304rposhbn/specifications.html)

메인보드는 S1200V3RPO

(https://www.intel.co.kr/content/www/kr/ko/products/sku/71386/intel-server-board-s1200v3rpo/specifications.html)

입니다.

 

CPU는 E3-1240L V3, MEM은 DDR3 2G * 4이며,

장착된 하드디스크는 2TB 4장, 80GB SSD1장 총 5장입니다.

이 중 2TB하드 3장은 proxmox에서 가상화로 돌렸던 헤놀로지에서 RAID5(SHR아님)로 돌렸던 디스크입니다.

별도의 PCI카드는 사용하지 않았으며, 메인보드의 SATA 0,1,2,3에 하드디스크 4개를 꽃았고 SATA5에 SSD를 꽃았습니다.

EUDEV, DDSML, EUDEV + DDSML 모두 시도해보았으며,

모델로는 DS920+, DS918+, DS3622xs+, SA6400을 시도했습니다.

부팅은 USB로 진행했습니다. 

 

증상은,

기존에 DSM에 사용하던 하드디스크임에도 불구하고 마이그레이션을 감지하지 않고 기기를 새로 설정하는 페이지가 뜹니다.
그런데 BTRFS로 이미 포맷되어 있는 디스크라 그런지 초기화를 요구하진 않고, 어쩌다가 한 번 DSM으로 진입하면 내부에 DATA는 모두 살아있음을 확인했습니다.
전원을 내렸다가 올린다던가 하는 식으로 재부팅을 하면 다시 기기설정페이지로 돌아갑니다.
DSM설치 후 재부팅하면 다시 기기설정페이지로 돌아가서 DSM을 무한재설치하게 됩니다.

이것을 계속 반복하고 있습니다.

혹시나 싶어 BIOS에서 AHCI -> Enhanced(Native SATA support)로 변경도 해보았는데.. 효과는 없었습니다.

 

특이사항(?)으로, 맨 처음 로더 빌드 후 DSM설치 -> 재부팅 후 모니터에 출력되는 것 중,

 

Detecting Upgrade : Ramdisk upgrade has been detected and sed : no previous ragexp

zImage upgrade has been detected

 

이렇게 두 줄이 출력됩니다.

신경을 안 썻던거였는진 몰라도.. 제 기억으로 DSM버전이 바뀌거나 하면

Detecting Upgrade : Ramdisk upgrade has been detected 요거 한 줄만 나왔던거같은데...

 

기타 그 외 메모리 오버 없습니다.

 

어떤 정보를 더 드려야 문제 해결이 될지 알려주시면 감사하겠습니다...

Fentanest, 2024-01-06:
[meesages2.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13850083/meesages2.txt)

Fentanest, 2024-01-06:
[messages3.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13850097/messages3.txt)
[messages2.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13850098/messages2.txt)

PeterSuh-Q3, 2024-01-06:
기존 하드에 설치된 DSM 버전과 새로 설치될 DSM 버전이 어떻게 달라서 어떻게 바뀌어야 하는지에 대한 정보가
linuxrc.syno.log 
파일의 맨 마지막에 나와 있습니다.

그런데, DS920+, SA6400 등의 Device-Tree 기반의 모델은 Device Mapping 처리와 관련된 내용을 너무 많이 로그에 뿌려서 로그가 후반부에 잘리는 현상이 있습니다.
나머지 두모델로 진행하신후 
linuxrc.syno.log 

를 첨부 부탁드립니다.

포럼에는 이 내용까지만 댓글 드리겠습니다.
여기 깃헙에서 계속 진행했으면 합니다.

Fentanest, 2024-01-07:
답변 감사합니다

[linuxrc.syno.log_DS3622xs+.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13851943/linuxrc.syno.log_DS3622xs%2B.txt)
[meesages_DS3622xs+.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13851944/meesages_DS3622xs%2B.txt)
[image]
[terminal.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13851945/terminal.txt)

DS3622xs+로 진행한 내용입니다

PeterSuh-Q3, 2024-01-07:
linuxrc.syno.log 는 이슈가 아닐것 같네요.
아마 DSM 설치가 처음한번은 정상 진행하셨을것 같구요.
재부팅 되면서 7.2.1 Update 3 로 올라가는 자동 램디스크 패치 과정에서 아래처럼 오류 메시지가 보이네요.
write error 가 나오는 정확한 오류원인은 모르겠지만, 3번째 파티션은 공간이 항상 충분해서 공간부족현상은 아닐것 같구요.

USB 스틱의 불량으로 의심됩니다.

Detecting upgrade : Ramdisk upgrade has been detected and od: write error
od: write error: Broken pipe
Hunk 1 FAILED 1/1.
-root:x:0:0::/root:/bin/ash
+root::0:0::/root:/bin/ash
 system:x:1:1::/usr/syno/synoman:/usr/bin/nologin
 daemon:x:2:2::/:/bin/sh
 SYSTEM_ADMIN:x:3:101::/nonexist:/usr/bin/nologin
Possibly reversed hunk 1 at 977
Hunk 1 FAILED 51/51.

Fentanest, 2024-01-07:
상기 에러는 이번에 처음 발생한 것이긴 합니다. 우선, USB메모리를 교체해서 한 번 진행해보겠습니다.

보통은 제가 위에 언급했던

Detecting Upgrade : Ramdisk upgrade has been detected and sed : no previous ragexp

zImage upgrade has been detected

이 문구가 출력되는 상태였습니다.

또한, 부팅을 USB뿐만이 아니라 80GB SSD를 활용해 SATA부팅으로도 시도했었는데.. 동일하게 무한 재설치 증상이 발현되었었습니다.

Fentanest, 2024-01-07:
혹시 이런 부분과는 무관한걸까요?

Found SATA Controller : pciid 8086d00008c02  Required Extension : Searching for matching extension for ahci
Found VGA Controller : pciid 102bd00000522  Required Extension : No matching extension
Found Ethernet Interface : pciid 8086d00001533 Required Extension : Searching for matching extension for igb
Module igb loaded succesfully
Found Ethernet Interface : pciid 8086d00001533 Required Extension : Searching for matching extension for igb
Module igb loaded succesfully
Found SATA Controller : pciid 1b4bd00009235  Required Extension : No matching extension
Found SATA Controller : pciid 1b4bd00009235  Required Extension : No matching extension

PeterSuh-Q3, 2024-01-07:
네, 정상메시지 입니다.

PeterSuh-Q3, 2024-01-07:
> 상기 에러는 이번에 처음 발생한 것이긴 합니다. 우선, USB메모리는 두 가지를 사용해봤었는데, 한 가지 더 USB메모리를 교체해서 한 번 진행해보겠습니다.
> 
> 보통은 제가 위에 언급했던
> 
> Detecting Upgrade : Ramdisk upgrade has been detected and sed : no previous ragexp
> 
> zImage upgrade has been detected
> 
> 이 문구가 출력되는 상태였습니다.
> 
> 또한, 부팅을 USB뿐만이 아니라 80GB SSD를 활용해 SATA부팅으로도 시도했었는데.. 동일하게 무한 재설치 증상이 발현되었었습니다.

보통 정상일 때는 

Detecting Upgrade : Ramdisk upgrade has been detected and sed : no previous ragexp zImage upgrade has been detected

가 나가는게 맞습니다.

SATA 보다는 USB가 더 안정적입니다.

Fentanest, 2024-01-07:
[linuxrc_DS918.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13853257/linuxrc_DS918.txt)
[messages_DS918.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/13853258/messages_DS918.txt)

USB를 교체한 후에도 동일한 증상이 발생하고 있습니다....ㅠㅠ

Fentanest, 2024-01-07:
[image]

이건 DS920+으로 시도하면서 보인 터미널이구요...ㅠ

Fentanest, 2024-01-07:
od write error 이 부분은 모니터에는 출력되지 않습니다.

모니터로는 Detecting Upgrade : Ramdisk upgrade has been detected and sed : no previous ragexp zImage upgrade has been detected 이 문구만 나오고,
브라우저에서 접근하면 od : write error가 보이는 상황입니다.

PeterSuh-Q3, 2024-01-07:
DS918+ 의 linuxrc 로그에 단서가 보이는것 같습니다.

Assemble args: -u 066de69c:d65142d2:3017a5a8:c86610be /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1 /dev/sdf1
mdadm: WARNING /dev/sdb1 and /dev/sdc1 appear to have very similar superblocks.
      If they are really different, please --zero the superblock on one
      If they are the same or overlap, please remove one from the list.
Exit on error [12] No raid status in path /sys/block/md0/md/array_state, go to junior mode...

->

인수 어셈블: -u 066de69c:d65142d2:3017a5a8:c86610be /dev/sda1 /dev/sdb1 /dev/sdc1 /dev/sdd1 /dev/sdf1
mdadm: 경고 /dev/sdb1 및 /dev/sdc1에는 매우 유사한 슈퍼블록이 있는 것 같습니다.
       정말 다르다면 하나의 슈퍼블록을 0으로 설정하세요.
       동일하거나 중복되는 경우 목록에서 하나를 삭제하세요.
오류 발생 시 종료 [12] /sys/block/md0/md/array_state 경로에 RAID 상태가 없습니다. 주니어 모드로 이동하세요...

정확히 어떤 디스크 3장이 RAID5 를 구성했었는지 말씀을 안해주셔서 추정만 하는데요.
로그의 순서대로 본다면

REVISION = #69057
hdmodel HDS723020BLA642 (sda)
fwrev MN6OA5C0
Append drive and firmware:
REVISION = #69057
hdmodel WD2003FYYS-02W0B1 (sdb)
fwrev 01.01D02
Append drive and firmware:
REVISION = #69057
hdmodel WD2003FYYS-02W0B1 (sdc)
fwrev 01.01D02
Insert firmware version:
REVISION = #69057
hdmodel WD20EZRX-00D8PB0 (sdd)
fwrev 80.00A80
Append drive and firmware:
REVISION = #69057
hdmodel SSDSC2CT080A4 (sdf)

이 순서가 맞는지 모르겠습니다.

RAID5로 구성된 디스크3장과 나머지 디스크 2장의 슈퍼블록이 따로 놀고 있는게 아닌가 싶습니다.

이 데이터들은 혹시 따로 백업은 되어 있나요?
백업이 없으면 모험이 될 수 밖에는 없지만,
일단 RAID5 그룹 전체3장 또는 나머지 디스크 2개중 1개만 가지고
마이그레이션을 먼져 해봤으면 합니다.

만약, 이렇게 마이그레이션이 한번이라도 성공하게 되면
정상 DSM이 로그인이 된 상태에서 나머지 디스크의 시스템 파티션의 DSM 도 자동 동기화로 따라가게 되어 있습니다.

이 슈퍼블록이 두 그룹으로 나귀면서 나머지 한그룹이 방해를 하고 있는게 아닌가 싶습니다.
이 방해에 의해서 DSM 로그인 진입을 못하고 계속 DSM 을 설치해야 하는 Junior 모드가 반복되고 있는것 같습니다.

PeterSuh-Q3, 2024-01-07:
마지막 메시지로 봐서는 이미 RAID5 구성이 깨져있는것 같기도 합니다.
( Exit on error [12] No raid status in path /sys/block/md0/md/array_state )

이 레이드5 디스크 3장은 제외하고 나머지 디스크로 먼저 진행해 보셔야 할것 같습니다.

Fentanest, 2024-01-08:
해당 메세지는 다른 모델로 진행할때도 출력되던 메세지라서
SHR이 아닌 Raid5로 구성해서 그런가보다. 했었습니다..ㅋㅋ

현재 RR로 로더 구성하여 부팅은 완료했고 데이터 풀 스크러빙 중입니다.
스크러빙하고 레이드가 무사하면 데이터를 빼낸 뒤 다시 m-shell로 시도해야 할 것 같습니다.

PeterSuh-Q3, 2024-01-08:
디스크는 5개 모두 그대로 두고 RR 로더로만 바꿔서 사용해 보신건가요?

Fentanest, 2024-01-08:
네 맞습니다. 맞는데 부팅해보니까 Raid5에서 디스크 하나가 충돌함 상태로 부팅이 되었고.. 지금은 고군분투중입니다.. 어쩌면... 다 잃을 수도 있겠네요 ㅜ

PeterSuh-Q3, 2024-01-08:
> 네 맞습니다. 맞는데 부팅해보니까 Raid5에서 디스크 하나가 충돌함 상태로 부팅이 되었고.. 지금은 고군분투중입니다.. 어쩌면... 다 잃을 수도 있겠네요 ㅜ

레이드5에서 1장은 충돌해도 나머지2장이 데이터를 보존하고 있기에 복구가능하실겁니다.

Fentanest, 2024-01-08:
한 장을 교체해주고 스크러빙까지 끝났는데도 충돌함 상태여서 지금 좀 많이 식겁합니다 ㅠ_ㅠ

PeterSuh-Q3, 2024-01-08:
서버포럼 자게에도 올리긴 했습니다만, 충돌난 하드는 두고 새로 한장더 스페어를 쓰셨다고 하셨는데요.
이 스페어가 물리적으로 문제가 있는건 아니겠죠?
보통은 충돌난 하드를 초기화해서 재구성에 다시 사용해도 됩니다.

Fentanest, 2024-01-08:
트러블슈팅과정에서 무언가 꼬였는지 충돌하드가 어느 하드인지 표시되어있지 않았습니다 ㅠ_ㅠ.
그래서 드라이브 1:1교체를 직접 지정하지 못하고 핫스페어 기능을 사용했습니다.

지금 현재는 충돌하드가 어느 하드인지까지 파악했고
일단 전체 확장s.m.a.r.t검사를 돌려놓고 있습니다

closed 2024-01-09 by PeterSuh-Q3