PeterSuh-Q3/tinycore-redpill #24: g7400cpu cannot load core display normally

zhuangzhuangY, 2024-01-14:
Sorry, my English is not very good, use Google Translate to carry on the conversation.
Problem: The hardware is g7400cpu. After installation, nothing can be obtained through dmesg | grep i915. The cpu model is the default epyc, I don’t know if it is affected.

PeterSuh-Q3, 2024-01-15:
I also get help from Google Translator.
I think your explanation is a bit lacking.
The g7400 cpu was used for transcoding, and it appears that the SA6400 model is being built and used in the loader.

Are there any abnormalities in the results of the two checks below?
dmesg | grep i915
ll /dev/dri

If H/W transcoding does not work even though the above results are normal,
There is a possibility that the video S/W settings such as Plex or Jellyfin were set incorrectly.
Please share the results from the above two commands first.

zhuangzhuangY, 2024-01-15:
thank you for your reply,
dmesg | grep i915 no output
ll /dev/dri log content: dm-0 dm-1, no nuclear display.
The RR boot is now installed and the result is consistent with your boot

zhuangzhuangY, 2024-01-15:
[image]
This is a screenshot of the log of the RR boot I am currently reading

zhuangzhuangY, 2024-01-15:
Good news for you, g7400 has been driven normally.
Cause: The motherboard can only call one graphics card. Because the independent graphics card is plugged into the machine, the core graphics cannot be used.

closed 2024-01-15 by zhuangzhuangY