PeterSuh-Q3/tinycore-redpill #27: i915 issue

decodering22, 2024-02-06:
After boot kernel panic and can't install DSM due to i915 issue. Is the i915 built into this latest build?

PeterSuh-Q3, 2024-02-06:
SA6400 ?

decodering22, 2024-02-06:
Hi Peter yes SA6400, been pulling my hair trying to get xpenology installed and no luck

PeterSuh-Q3, 2024-02-06:
BareMetal or VM like Proxmox?

decodering22, 2024-02-06:
Baremetal intel i5-12500 on MSI MGP 760I motherboard. If there is anything you want me to try let me know I'm in front of it now, been battling this for a few hours

decodering22, 2024-02-06:
I only tried ARC and RR. Initialy i tried RR and it just kernel panic so assumption is i915 is the issue but I'm not sure. I tried your RR build and after the QR code pops up then screen goes blank. I assumed based on that thread the latest i915 was loaded on i915 so it should work. Not sure what it could be b/c there is no DSM log since I can't even install DSM. 

do you want to email or even call me?

PeterSuh-Q3, 2024-02-06:
https://xpenology.com/forum/topic/60130-redpill-tinycore-loader-installation-guide-for-dsm-71-baremetal/?do=findComment&comment=456343

Looking at this log, it appears that mshell has already been tried.

Are you sure that i915 is causing the panic by checking the logs?
It's hard to be sure until you check dmesg yourself.
rr and mshell provide functions to retrieve and analyze dmesg from error situations.

The fact that the screen disappears on SA6400 may mean that i915.ko is working well.
The bug where the screen disappears after i915.ko is loaded properly is also confirmed in mshell.

It seems to me that the Realtek 2.5G network is not operating stably.
Are there any additional NICs that can be installed? It's better if it's from Intel.

decodering22, 2024-02-06:
I have a 10Gbe intel card i plan to use you think if i disable the onboard Realtek that will fix the issue? I can disable it and test it for sure that is not a problem i will disable it on the UEFI bios. 

I can't check the log b/c as soon as the IP loads it causes the kernel panic. The only reason i went down the i915 rabbit hole is b/c it was mentioned that was the issue. Let me pop in the intel X520-DAC1 and disable the Realtek 2.5G to see if that works or not. 

Where can i pull the log to be 100% sure?

decodering22, 2024-02-06:
I also have a 1GBe intel card i will pop in for the test as well

decodering22, 2024-02-06:
Same thing, i disabled the Realtek 2.5Gbe NIC in the BIOS itself, it detected the new pcie 1Gbe Intel VRRH1-HIGH Nic without issues. and the result is the same. Screen goes black and after about 15 seconds I can ping it again but can't be found in finds.synology.com or the IP with port to connect to it. 

Can you tell me where to pull the log from and when. Really appreciate your help

decodering22, 2024-02-06:
If there is some settings you want me to try and test let me know. Not sure how to pull the log b/c it kernel panics right away

PeterSuh-Q3, 2024-02-06:
In the case of mshell, the last log file is saved when the Friend kernel is loaded.
It exists in the directory below.
/mnt/tcrp-p1/logs/jr
Friend, stop with ctrl+c while booting.
Please examine the files under this directory.
You should also look at dmesg, but the contents of linuxrc.syno.log are also important.
Can you provide me with these 2 files for download?

[image]

decodering22, 2024-02-06:
Yes let me load RR on a flash drive right now and get these to you now. Give me a min I have to go through the entire process probably a few mins to get it to the point where it does this. Do you want me to email it to you directly or do you want to continue here on github?

PeterSuh-Q3, 2024-02-06:
I gave an example of MSHELL. I don't know exactly where and how RR's logs are stored.
GitHub is also good enough.

decodering22, 2024-02-06:
I have your tinycore-redpill that is what I meant when i said RR not sure if your reference to RR is something else but i assume the redpill is the MSHELL which is what I loaded prior and now

decodering22, 2024-02-06:
I'm following this procedure although at the end it doesn't exactly work per the instructions
https://xpenology.club/install-dsm-7-on-baremetal-or-vm/

decodering22, 2024-02-06:
Ok at step Build the Loader 7.2.1.-69057, Friend) and about to reboot

decodering22, 2024-02-06:
Ok just rebooted 
[image]

decodering22, 2024-02-06:
Ok just went to blank screen, I'm going to pull the logs like you said and post it
[image]

decodering22, 2024-02-06:
Peter there is no log in the /mnt/tcrp-p1# folder

decodering22, 2024-02-06:
what is the user and pass default so i can use WinSCP to transfer it over and attach it here?

decodering22, 2024-02-06:
I can keep trying ot make it crash to make it have a log but its not storing a log. 

[image]

decodering22, 2024-02-06:
Need to figure out how to export the dmesg so i can copy it over. Was hoping to use WinSCP but i don't know the user and password as its not tc & P@ssw0rd anymore

decodering22, 2024-02-06:
Peter how do i get the dmesg i saved to a txt file off so i can transfer it over to windows to send it to you?

decodering22, 2024-02-06:
Ok i was able to connect using the PC and ran the dmesg and copied and paste from the Browser. Let me know what else I can provide. I don't see the logs in the other foler?

[dmesg.txt](https://github.com/PeterSuh-Q3/tinycore-redpill/files/14174973/dmesg.txt)

decodering22, 2024-02-06:
shouldn't there be some reference to i915 in dmesg?

decodering22, 2024-02-06:
Anything else you want me to grab?

decodering22, 2024-02-06:
Another note after the screen goes blank it takes down the NIC with it completely likely due to kernel panic for what its worth. 

Let me know your thoughts after looking at the dmesg output. there is no linuxrc.syno.log b/c DSM install never got started b/c of the kernel panic

decodering22, 2024-02-06:
Calling it a night Peter tomorrow if you have any idea let me know. If there is any specific BIOS settings you think could be it let me know. I'm willing to try everything, spend a good bit of money on this and if it doesn't work its all for nothing. thanks for your help

PeterSuh-Q3, 2024-02-06:
I'm sorry. I didn't have time to check GitHub because I was busy with work.

I don't know if you understand this script, but
This is the core content that contains the log.
I don't think I had a chance to copy the logs from the two copies - the earliest and the latest when Junior boots.

There are two more events in between. (There are two types: jrExit and modules.)
We will improve the dbgutils Addon so that it can retrieve logs from these two events.

We will share this addon again when it is more complete.

https://github.com/PeterSuh-Q3/tcrp-addons/blob/main/dbgutils/src/install.sh

decodering22, 2024-02-06:
Hi Peter, do you want me to manually run this script? Just trying to understand what I can do now to help isolate the problem and get this fixed?

decodering22, 2024-02-06:
Did you see the dmseg output i uploaded? Anything else I can do now to fix this

PeterSuh-Q3, 2024-02-06:
This page has already been changed and you cannot do this yourself. Logging out is one of Mosaic's high-level improvement tasks. I have already tried this several times and failed. I'll try it one more time and after a few days, I hope you don't rush and give me time.

PeterSuh-Q3, 2024-02-06:
The most stable and efficient way to view logs is to monitor console output through the serial com port. However, since this device is no longer supported in the new MOBO, we did not mention this from the beginning. It is also easy to check for kernel panic or NIC unresponsiveness here. There is something not mentioned about how to check IP. The IP displayed on the FRIEND screen may not be 100% reliable. So as a workaround, I use finds.synology.com and if it doesn't show up, I look into the IP assigned to my router. The best way is to fix the IP by pre-setting the Mac address you know to the router. Try securing an IP address using this method first.

decodering22, 2024-02-06:
Hi Peter, i have both finds.synology.com up in a browser and I'm pinging and using the browser to connect, all 3 at the same time. I can confirm when the "kernel panic" occurs it drops the IP all together and the finds.synology.com i'm refresing the whole time never finds the DSM. You are right there is no more serial port. Is there any other way while you work on that script i can try different settnigs or options you can think of?

PeterSuh-Q3, 2024-02-06:
Analysis of logs is currently the best method; anything else can only be guesswork. You keep mentioning kernel panics, but even in non-kernel panic situations, the loader can crash with an error. And it seems that any hardware option adjustments will no longer be effective. Redfill models that can operate in the 12th generation are limited, and the SA6400 is a model that has just entered stabilization, so more users' experience is needed. Other than my Mshell, another option may be the rr loader from a developer called wjz304. I know that the sharing of this loader has been closed, but if there is a way to do so, I would like to obtain it and try it.

decodering22, 2024-02-06:
Sure what ever you want me to try i will do it, just let me know. The sooner you can get me the script to look at the logs the better. Just let me know what you want me to try. Are you are correct it could be other issue causing it. Just need to narrow it down

decodering22, 2024-02-06:
Peter was the dmseg output of any value when you looked at it?

decodering22, 2024-02-06:
I just tried this ARPL i think it is by wjz304, ti does the same it hangs. but I don't think there is anyway to pull any logs from it?
https://github.com/fbelavenuto/arpl/releases/tag/v1.1-beta2a

decodering22, 2024-02-06:
Peter I also want to mention, i have no problem paying you for your time helping me. I can paypal you just let me know. Thanks again

decodering22, 2024-02-06:
you are correct i can't get access to this repo https://github.com/wjz304/arpl-i18n/releases 

If you have it i can try it

PeterSuh-Q3, 2024-02-07:
Use this
https://github.com/syno-community/rr/releases

decodering22, 2024-02-07:
Thanks Peter i will try this as well. I ordered a new motherboard as well and it should be here today. I took the other system apart already so when the new one arrives I will try this on it.

PeterSuh-Q3, 2024-02-07:
These are successful cases using Intel 12th generation and SA6400, including N100, at the server forum in Korea where I work. It would have been better if you had let me know in advance.
good luck.

https://svrforum.com/?_filter=search&act=&vid=&mid=case&category=&search_target=extra_vars5&search_keyword=12

decodering22, 2024-02-07:
Thanks Peter I'll translate the site and see. It has to be some compatibility with the Motherborad i was using. I'm still waiting for the new one to arrive, it does not have any Realtek NIC, only Intel, but still not sure if that was even the issue since we couldn't capture logs before it crashes. 

I'll let you know as soon as I have the new system to see if it works.

decodering22, 2024-02-09:
Hi Peter, so I had some luck with this new motherboard and it is working but an odd issue is if i remove the USB drive it just goes straight to the BIOS. It seems like i have to have the USB drive plugged in all the time so it forces DSM to boot. Is this normal? Is there a way to make the SATA drive bootable so no USB is needed?

decodering22, 2024-02-09:
Peter just so i understand the loader needs a USB drive to boot, than it installs the DSM on the SATA SSD correct. Is it possible to partition part of the SATA SSD so it is used to boot the DSM and act like the loader?

PeterSuh-Q3, 2024-02-10:
It's impossible. You need to understand the mechanism by which Synology disks' boot devices and data disks work. All data disks have the same system partition at the front of the disk, and DSM is installed equally there. DSM itself does not support booting. Genuine devices required for Synology booting use durable Sata Dom. Redpill loaders such as TCRP and ARPL create boot loaders with the same structure as this sata dom. Small capacity USB, MSata, SSD, etc. must be made independent with a separate bootloader. M.2 Sata SSD is also available, but Nvme is not available.

decodering22, 2024-02-10:
If i understand the USB or SATA DOM is needed no matter what. The SATA disk and M.2 based on what you stated both have the same DSM system partition. So if I add a second M.2 drive its not just a Volume drive only, but there is a small system partition as well correct?  I'm going to just use a small low profile and fast USB and call it a day. I was hoping to avoid it but it doesn't sound like it is possible. 

It is only used during boot only correct, so minimal Read/Write on the USB? I don't reboot often so I suspect it should be fine?

PeterSuh-Q3, 2024-02-10:
There are two types of M.2, so they must be clearly distinguished.
These are M.2 Sata SSD and M.2 NVMe SSD.
If you add an M.2 Sata SSD, it operates the same as a regular SATA (SSD/HDD) disk.
Any Sata based disk can be used as a bootloader.
However, M.2 NVMe SSD can only exist as a cache, but with the help of a special addon, it can be turned into a data disk volume. It cannot be used as a bootloader.

USB sticks are the most vulnerable devices in terms of durability.
It stays on the PC during booting and does not participate in reading/writing, but
Must be exposed to continuous heat.
This is a factor that reduces the durability of the USB stick.

decodering22, 2024-02-10:
Hi Peter, great explanataion, I know the difference between the two types of M.2 drives and how they are utilized in xpenology pradigm with m.2 as cache but with the drive DB update being able to be used as data disk volume. 

Tracking that M.2 NVMe PCIe can't be used for bootloader and right now it isn't fully supported without a SATA drive. 

What I was concerned with was the USB drive being suseptiple to failure and if it was possible to partition the SATA SSD drive for a boot partition and the other partition being used for the DSM System partition. It sounds like that is not possible at the moment and the use of a USB or separate SSD(SATA) or SATA DOM is needed for the boot loader alone. 

I was under the impression with a previous version (Jun's) release in DSM 6.x days he had figured out a way to use the SSD (SATA) as both a bootloader and DSM system partition too. I just assumed that was also possible now with DSM 7.2 and beyond. 

In your opinion is it worth it to transition to a SATA DOM for long term reliability over USB?

PeterSuh-Q3, 2024-02-10:
I've seen users using SATA Doms instead of actual USB disks. I haven't directly encountered verified data on durability, but isn't the fact that Synology is using it enough verification?

decodering22, 2024-02-10:
Yea if Synology is using a SATA DOM than it must be reliable enough. I don't mind getting a small cheap SATA drive. I'm just concerned the USB disk failing overtime like you mentioned due to heat. The only options for SATA DOM are supermicro i have found and they don't have the greatest review. I have plenty of SSD lying around and there is plenty of open SATA slots. For now I ordered a Sandisk Fit and will use that until it fails than transfer to another option. Thanks for your help