PeterSuh-Q3/tinycore-redpill #28: Support for DS218+

antno1000, 2024-02-06:
Can you please add support for DS218+

PeterSuh-Q3, 2024-02-07:
OK, I'll support DS218+

CuriousLad1000, 2024-02-12:
Hello, Can you please make a bootable img release so I can use it and test.

antno1000, 2024-02-12:
@PeterSuh-Q3 Can you please make a img release of the current changes. I would like to try it out.

PeterSuh-Q3, 2024-02-12:
When you re-enter the mshell loader build menu, it will change to support DS218+ with automatic updates. No need for a new img. However, my testing has not yet confirmed a stable condition.

antno1000, 2024-02-29:
Hello, I was able to build the loader for DS218+ however, I am unable to get the DSM first setup screen either through the IP address or from finds.synology.com

[image]

PeterSuh-Q3, 2024-02-29:
As I mentioned above, I have not yet achieved a steady state.
As a result of checking the log coming out through the serial port in the background, 
the kernel panic continues.
There doesn't seem to be a solution right now.

antno1000, 2024-02-29:
So, for now there's no functional configuration for this model. Can I expect something in the future releases?

PeterSuh-Q3, 2024-03-04:
Nothing can be guaranteed.

PeterSuh-Q3, 2024-04-11:
Today I rebuilt lkm for DS218+, which has been missing since mid-March.
I then conducted a DSM installation test and confirmed normal installation as shown below.

[image]

closed 2024-04-15 by PeterSuh-Q3