PeterSuh-Q3/tinycore-redpill #30: [DS918+] iSCSI 관련 이슈입니다.

yymin1022, 2024-02-24:


PeterSuh-Q3, 2024-02-25:
당시 1월초에 MAC-SPOOF 를 강제 적용한 문제가 있던 버전을 설치 하셨던것 같고 당시 SAN MAMAGER (iSCSI) 를 
손상시키는 문제가 있었던 버전이 동작 되었던것 같습니다.
한번 손상된 SAN MAMAGER (iSCSI)는 현재로서는 복구가 불가능합니다.

저도 아직 iSCSI 에 대한 충분한 지식이 없어서
san manager 서비스만 간신히 기동해서 vmm 만 동작하도록 만든 상태입니다.
더 깊은 내용을 아직 모르기에 san manager 가 완벽하게 복구 동작되도록 구현을 못드리고 있는 실정입니다.

지금은 문제가 해소된 버전을 처음부터 한번더 클린설치를 하시던가
아니면 시놀로지 안에서의 설정 초기화 방법 밖에는 없는것 같습니다.

설정 초기화로 진행하신다면 데이터 파티션은 건드려지면 안됩니다.
이떄는 DSM 이 설치되었던 시스템 파티션만 초기화가 되어야 합니다.

yymin1022, 2024-02-25:
@PeterSuh-Q3
그렇다면 지금 DSM이 설치되어있는 상태에서, TCRP로 부팅하여 부트로더 업데이트만 진행하는 것은 의미가 없고,
새로 USB와 HDD를 포맷하며 DSM 재설치 후 데이터를 복원하면 사용이 가능하다는 것으로 이해하면 될까요?

PeterSuh-Q3, 2024-02-25:
네, 현재는 당시 문제를 일으켰던 MAC-SPOOF ADDON 은 선택적 설치방식으로 바뀌여서 기본선택되어 들어가지 않도록 되어 있습니다.
새로 셋팅하시면 관련 문제가 없어야 합니다.

yymin1022, 2024-02-25:
@PeterSuh-Q3
답변 감사드립니다.
혹시 해당 Add-On이 선택 설치로 변경된 시점이 언제인지 알 수 있을까요?

PeterSuh-Q3, 2024-02-25:
이 시점 입니다.
https://svrforum.com/nas/1201762

yymin1022, 2024-02-25:
@PeterSuh-Q3 
확인 감사드립니다.
제가 DSM을 최종적으로 재설치한 날짜는 1월 28일 경인데, 2월 2일에 패치가 배포된 것이군요.
그럼 이번에 DSM을 클린설치하면 이슈가 해결될 것으로 예상되네요.
주중에 시도해보고, 해결되는대로 Issue Close 처리하겠습니다!

yymin1022, 2024-02-27:
TCRP로 부팅해서 최신버전으로 업데이트 시킨 뒤, Grub 메뉴에서 DSM 재설치로 부팅해서 설정/파일 유지 옵션으로 DSM 자체만 재설치 했더니 해결되었습니다.
감사합니다!

yymin1022, 2024-03-07:
@PeterSuh-Q3 
지난 번 DSM 재설치로 iSCSI 이슈가 해결되었으나, DSM 재시작 시 동일증상과 함께 iSCSI 사용이 불가합니다.
아예 HDD 자체를 초기화하고 전체 클린설치를 해도 동일하며, 최초에 DSM 설치를 동반한 부팅시에만 iSCSI가 사용 가능합니다.
이후에 한번이라도 재부팅을 하는 경우, DSM 재설치를 수행해야만 iSCSI가 다시 사용 가능합니다.
다만, 이전에 발생하던 SAN Manager 오류 알림은 따로 발생하지는 않고, 단지 iSCSI만 사용이 불가한 증상입니다.