PeterSuh-Q3/tinycore-redpill #38: bootloader로 nvme를 사용할때 문제가 생긴것 같습니다.

nanalynKR, 2024-07-04:
안녕하세요.
항상 감사한 마음으로 잘 사용하고 있습니다.
esxi 환경에서 0.1.1c때부터 부트로더를 nvme로 사용하고 있었습니다.
최근에 nic 순서를 바로잡아주는 기능이 생겼길래 테스트용으로 하나 만들어서 시험해보고있는데
로더 빌드를 정상적으로 끝내고 나서도 grub 선택창에 맨처음 빌드 로더 하나밖에 나오지 않고있습니다.
빌드하는 화면에서도 grub 정보를 잘 못긁어 오는것 같습니다.
[image]

마지막에 추가된 nvmesystem addon은 적용하지 않았습니다.
해결 방안이 있을까요?

PeterSuh-Q3, 2024-07-05:
> 안녕하세요. 항상 감사한 마음으로 잘 사용하고 있습니다. esxi 환경에서 0.1.1c때부터 부트로더를 nvme로 사용하고 있었습니다. 최근에 nic 순서를 바로잡아주는 기능이 생겼길래 테스트용으로 하나 만들어서 시험해보고있는데 로더 빌드를 정상적으로 끝내고 나서도 grub 선택창에 맨처음 빌드 로더 하나밖에 나오지 않고있습니다. 빌드하는 화면에서도 grub 정보를 잘 못긁어 오는것 같습니다. ![2](https://private-user-images.githubusercontent.com/109270188/345797888-d4f6b5d1-580b-487b-a619-443f4f17e27c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxNzcxMDIsIm5iZiI6MTcyMDE3NjgwMiwicGF0aCI6Ii8xMDkyNzAxODgvMzQ1Nzk3ODg4LWQ0ZjZiNWQxLTU4MGItNDg3Yi1hNjE5LTQ0M2Y0ZjE3ZTI3Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwNVQxMDUzMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZGVmN2FmYWZhOWM1M2NkYjY1YzJiZmQ3NTg1Yjg4OWZkYjJlZmI0YjU5YTc2ZjU0N2VhOTc2YWUxYWY1OWIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.KdxPVVS6m2R6bBSVyxVmNJokF52HteXsz06FyocjrJk)
> 
> 마지막에 추가된 nvmesystem addon은 적용하지 않았습니다. 해결 방안이 있을까요?

안녕하세요.
esxi 가상환경만 문제가 아닌 네이티브까지 포함한 전반적인 오류라고 염두해 두고 재시험 해보도록 하겠습니다.
결과 조금뒤 피드백 드리겠습니다.

PeterSuh-Q3, 2024-07-05:
NVMe 의 경로는 USB 와 좀 다른 경로를 가지고 있는데요.
경로중 n1p1/n1p2/n1p3 를 n11/n12/n13 이렇게 p를 빼먹고 잘못사용하고 있었습니다.
해당 사항은 아래와 같이 수정했습니다. 단순표기 오류라 버전업은 하지 않았습니다.

https://github.com/PeterSuh-Q3/tinycore-redpill/commit/aefbd1d6a1f88577dc0a791c3d0dce6b8c1a8425

로더를 재빌드하거나 아래커맨드로 이 파일을 다운로드후 백업한번하시고 재부팅하시면 우측상단에 NVMe 경로가 제대로 표시되실겁니다.

curl -kLO https://raw.githubusercontent.com/PeterSuh-Q3/tinycore-redpill/master/functions.sh

단순표기 오류라서 로더 빌드에는 영향이 없습니다.

버그리포트 감사합니다.

closed 2024-07-05 by PeterSuh-Q3