PeterSuh-Q3/tinycore-redpill #40: 1.0.4.3 issues.

XBGT351, 2024-09-04:
HP N40L Microserver with modded bios.

1.0.4.3 seems to have a couple of issues. 
Issue 1. 
After redpill setup and first reboot, it loads the kernel, but there's no network access. ( using a dhcp MAC reservation )
Can't access synology setup wizard as the device is non responsive to ping.      Re-imaged several times with same result.

Issue 2.
After configuring redpill and rebooting, any further attempts to go back into the redpill setup wizard to change settings results in booting into tinycore without any setup wizard.  

For reference, 1.0.4.2 didn't have the above issues, but had a different issue of unable to upload or download 7.2.2 pat during the Synology setup.

PeterSuh-Q3, 2024-09-09:
There was an instability in the lkm module between v1.0.4.3 and v1.0.4.4. 
Kernel panics were present and this KP state can appear as if the NIC is unresponsive. 
Please try again with the latest stable version, v1.0.4.6, and let us know the results.

XBGT351, 2024-09-09:
1.04.6 running nicely.   
Perfect!  7.2.2 running in just a few minutes.

closed 2024-09-10 by PeterSuh-Q3