PeterSuh-Q3/tinycore-redpill #41: Sataportmap,DiskIdxMap to blank for VIRTUAL MACHINE

devz3ro, 2024-09-06:
Is it possible for you to change the routine in your script so that if the hardware is detected to be "VIRTUAL MACHINE" that the "Sataportmap,DiskIdxMap" values aren't automatically set to blank? I am trying to set the Sataportmap to 11 and DiskIdxMap to 1010 which is what I had it set to before using the original tcrp [pocopico] build and it was working as expected. With these values set to blank, I can't see 2 of my drives connected to the HBA inside of DSM.

I've tried manually editing the "user_config.json" to add the values and even editing in the values from the menu. During the build process they keep getting reset to blank.

The sata_remap option in the menu doesn't help, I am still unable to see 2 drives after running it + rebuild.

Here is the output from the "zlastbuild.log" file"

Machine is VIRTUAL Hypervisor=KVM
Found SAS HBAs, Restrict use of DT Models.
Sataportmap,DiskIdxMap to blank for VIRTUAL MACHINE

I tried setting the CPU type to "host" but tcrp still recognizes the VM as a Virtual Machine and blanks my manually added values before building every time.

devz3ro, 2024-09-06:
I was able to solve this myself by making 2 changes to the VM's config file in /etc/pve/qemu-server

Changed the following:
`cpu: x86-64-v2-AES`
to
`cpu: host`

Added the following to the first line:
`args: -cpu host,-hypervisor -device qemu-xhci,id=usb -device usb-tablet`

Then I was able to change the Sataportmap,DiskIdxMap fields and build without them being blanked out. I am now able to see all disks.

PeterSuh-Q3, 2024-09-09:
The features you want are already prepared. Please use the menu below before building the loader, and then build the loader.

[image]

[image]

closed 2024-09-09 by PeterSuh-Q3