PeterSuh-Q3/tinycore-redpill #75: RTL8126 Driver 추가 좀 해주세요

dateno1, 2024-11-22:
커널 모듈 목록 확인해도 r(tl)8126은 아직 없는것같습니다

최근 8125랑 거의 동일한 가격에 8126을 판매하고 있어서 신규 구매하게 된다면 이것만 사지 8125를 더 이상 사진 않을것같습니다

8125 모듈 컴파일시 8126도 같이 컴파일해서 추가 부탁드립니다

모듈 추가후 로더에 반영해주시면 (자동 업데이트니 재부팅만 하면 알아서  최신 올라갈꺼라 봅니다) 직접 시험해볼 작정입니다 (사무실에 5G SW가 없는 관계로 시험되는건 2.5G환경입니다)

PeterSuh-Q3, 2024-11-23:
지금 사용하고 계신 SA6400 은 기본적으로 rr 의 모듈을 그대로 사용하고 있습니다.
이미 r8126 모듈이 포함되어 있음을 확인했습니다.

[image]

혹시 SA6400 외 다른모델로 변경 사용을 원하시면 
mshell 의 all-modules 가 아닌 rr-modules 로 전환할 수 있는 기능이 최근에 개발되었습니다.
이 공지글을 한번 보고 따라가 주시기 바랍니다.

https://svrforum.com/nas/1868979

closed 2024-11-23 by PeterSuh-Q3

dateno1, 2024-11-24:
> 지금 사용하고 계신 SA6400 은 기본적으로 rr 의 모듈을 그대로 사용하고 있습니다. 이미 r8126 모듈이 포함되어 있음을 확인했습니다.
> 
> ![스크린샷 2024-11-23 오후 3 25 33](https://private-user-images.githubusercontent.com/85427533/389178010-49a198a3-b6ea-4e39-adf5-4f7111d415a2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI0MDgyOTcsIm5iZiI6MTczMjQwNzk5NywicGF0aCI6Ii84NTQyNzUzMy8zODkxNzgwMTAtNDlhMTk4YTMtYjZlYS00ZTM5LWFkZjUtNGY3MTExZDQxNWEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI0VDAwMjYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVkMjEyMTY3ZGFkMWQyY2M0ZWIxZjg5NWI0NzJlOWRmYTUwYjhjMGFkYjBlNGJlYzgyNTFkMzkyNzU3NTg3OTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.hm7NSEWdr9U6SleJlCcRs56CEDE5X87Gwe1RpNst8Ts)
> 
> 혹시 SA6400 외 다른모델로 변경 사용을 원하시면 mshell 의 all-modules 가 아닌 rr-modules 로 전환할 수 있는 기능이 최근에 개발되었습니다. 이 공지글을 한번 보고 따라가 주시기 바랍니다.
> 
> https://svrforum.com/nas/1868979

시험에 사용한 기종은 SA6400이 아닌 DS3622xs입니다

lspci -v 했을때 모듈 안 떳고, DSM에서 NIC 안 잡혔습니다