PeterSuh-Q3/tinycore-redpill #77: HBA disk order

Nick1100, 2024-11-28:
I have a server with 6 SATA ports onboard and an LSI 8-port HBA in IT mode. In DSM, I see disk 1 through 6 correctly. Disk slot 7 is skipped, and the LSI HBA counts from disk slot 8 in a random order.

Another server has a 16-port LSI HBA, and on SAS8087 ports 3 and 4, there is a SAS expander to 4x SAS 8087. In total, I have 24x 2TB HDDs. With TrueNAS, I see all the HDDs. In DSM, I see a maximum of 16 HDDs, even though I have set maxdisk to 24.

What settings do I need in sataportmap or similar? I have been searching and trying for days without success. Help?

kenny-nt, 2024-12-02:
My DS923+ must use sata_remap to make disks have right order again. But I must set SataPortMap and DiskIdxMap to "" or loader build process will be fail.

kokofixcomputers, 2024-12-15:
> sata_remap

Hi there! I'm new to the configuration file. Can you tell me how to use `sata_remap`?

PeterSuh-Q3, 2024-12-16:
> > sata_remap
> 
> Hi there! I'm new to the configuration file. Can you tell me how to use `sata_remap`?

Please see this notice.

https://xpenology.com/forum/topic/61839-tinycore-redpill-loader-build-support-tool-m-shell/?do=findComment&comment=465702

kenny-nt, 2024-12-16:
> > > sata_remap
> > 
> > Hi there! I'm new to the configuration file. Can you tell me how to use `sata_remap`?
> 
> 
> Please see this notice.
> 
> https://xpenology.com/forum/topic/61839-tinycore-redpill-loader-build-support-tool-m-shell/?do=findComment&comment=465702

Use that option in VMWare > Working Good.
Select that option on J1900 > Menu exist with error. Don't remember error details but something like SATA_PORT[arg]

closed 2025-05-13 by PeterSuh-Q3