PeterSuh-Q3/tinycore-redpill #81: DS3615xs Does not build.

kokofixcomputers, 2024-12-21:
Trying to build DS3615xs causes: "Checksum mismatch - expected  but computed <checksum>" and i think why is because the sha256 field in here: https://github.com/PeterSuh-Q3/tcrp-modules/blob/main/all-modules/releases/bromolow.json is left blank.

PeterSuh-Q3, 2024-12-21:
I just modified it and took action.

PeterSuh-Q3, 2024-12-22:
Thank you for your help in correcting the word. I have applied it to functions.sh.
I modified sha256 and the build is fine, but I checked that KP is still running. (VMWARE FUSION environment).
Please wait until the problem is resolved.

PeterSuh-Q3, 2024-12-22:
Like RR or ARC, mshell should also stop supporting DS3615xs.
It is burdensome to continue this model that stopped supporting Kernel 3.
I think it stopped for the same reason as both loaders.

It seems not worth investing time, resources, and effort.

Since pocopico's TCRP is stopped in the old version, I thought DS3615xs would work, but KP occurs again as shown in this capture.

I recommend migrating to DS3622xs+ instead of the successor model DS3615xs.
As of today, DS3615xs is stopped with version 1.0.6.6.

[image]

closed 2024-12-22 by PeterSuh-Q3

kokofixcomputers, 2025-01-02:
Thank you for your time and support for maintaining the project. I will move to DS3622xs+ and try again. Thanks for trying to help!

PeterSuh-Q3, 2025-02-18:
DS3615xs is supported again from version 1.2.1.6. 
Please use DSM version 7.1.1.