PeterSuh-Q3/tinycore-redpill #101: mmc loader disabled with hard-wired synoboot_satadom=0 [DISABLE] on eMMC

RSfuchs, 2025-05-02:
it doesn't work for me with loader type mmc on eMMC as DS925+. During normal boot, getting "Invalid Loader Type. Exiting!" in tcrpfriend kernel.

synoboot_satadom=0 ?
need: "Active Fake Satadom Option" !

Doesn't my Synoboot USB stick need because of its VID=0xF400 PID=0xF400 also: synoboot_satadom=1 ?

PeterSuh-Q3, 2025-05-11:
I will provide a function that can record 3 option values ​​0, 1, 2 like rr's menu method. 

This menu should be used after loader build.

PeterSuh-Q3, 2025-05-12:
The menu function promised in the morning has been implemented and distributed.
0 (Disable) / 1 (Native) / 2 (Fake)
You can switch between three values, and you don't need to rebuild the loader.

As mentioned in the XPE community, this is a function for sata_line.
It does not apply to mmc that uses usb_line.

I hope the XPE community will continue the detailed story.
Can we close this issue for now?

https://github.com/PeterSuh-Q3/tinycore-redpill/blob/main/menu_m.sh#L1400

RSfuchs, 2025-05-12:
The changes are immediately visible, but the boot screen still displays "Cmdline: ....synoboot_satadom=0."

Now it's stuck in a loop during boot time:
No Redpill loader partitions found. Exiting!!!
Wait for additional time until mmc device is recognized...
mmc device not detected after waiting.
No Redpill loader partitions found. Exiting!!!
Wait for additional time until mmc device is recognized...
mmc device not detected after waiting.
No Redpill loader partitions found. Exiting!!!
Wait for additional time until mmc device is recognized...
mmc device not detected after waiting.
No Redpill loader partitions found. Exiting!!!
Wait for additional time until mmc device is recognized...

....

RSfuchs, 2025-05-14:
On 4/30/2025 at 2:03 AM, [Peter Suh](https://xpenology.com/forum/?app=core&module=members&controller=profile&id=46964) said:
However, I found out that Kernel 5 requires the setting of synoboot_satadom=0 [DISABLE].

This realization rendered the loader unusable.

RSfuchs, 2025-05-14:
I can confirm the need to set the Fake Sata DoM option in the USB line for eMMC Loader, too.
Just activating it in the menu is not enough. Otherwise the loader is running again.

RSfuchs, 2025-06-12:
Thank you, the problem has been resolved and could be tested without any problems as a Kernel 5 model DS425+.